HARPLab/DReyeVR #93: What does data structure mean in DReyeVR sensor data?

wx983, 2023-03-20:
I can understand  what other data means, but I can't understand them:

[image]
[image]
I can understand  what other data means, but I can't understand them:

GustavoSilvera, 2023-03-20:
Sorry for the lacking explanations. We'll be working on documentation soon.

The `FocusInfo` tracks a struct containing the following (in [DReyeVRData.h](https://github.com/HARPLab/DReyeVR/blob/70a15b5b05b135682cc2daf8ddc4f469e9f9f05c/Carla/Sensor/DReyeVRData.h#L97-L103)):
```c++
TWeakObjectPtr<class AActor> Actor; // pointer to UE4 actor (for internal use)
FVector HitPoint;                   // world-space location of the location of the ray-actor intersection
FVector Normal;                     // normal of the surface at the ray-actor intersection
FString ActorNameTag = "None";      // Tag of the actor being focused on
float Distance;                     // Distance (in meters) the ray traveled until intersecting the actor
bool bDidHit;                       // Whether or not an actor was intersected with the gaze ray
```

And For the EgoVariables we have (also in [DReyeVRData.h](https://github.com/HARPLab/DReyeVR/blob/70a15b5b05b135682cc2daf8ddc4f469e9f9f05c/Carla/Sensor/DReyeVRData.h#L61-L69)):
```c++
// World coordinate Ego vehicle location & rotation
FVector VehicleLocation = FVector::ZeroVector;
FRotator VehicleRotation = FRotator::ZeroRotator;
// Relative Camera position and orientation (for HMD offsets)
FVector CameraLocation = FVector::ZeroVector;
FRotator CameraRotation = FRotator::ZeroRotator;
// Absolute Camera position and orientation (includes vehicle & HMD offset)
FVector CameraLocationAbs = FVector::ZeroVector;
FRotator CameraRotationAbs = FRotator::ZeroRotator;
```

wx983, 2023-03-21:
@GustavoSilvera OK, I see, but what HMD refers to? Is it the VR helmet?

wx983, 2023-03-21:
@GustavoSilvera If I want to see these tags, should I change them to `True`? In DReyeVR, they are `false` in default.
[image]

GustavoSilvera, 2023-03-21:
Yes HMD stands for ["Head Mounted Display"](https://en.wikipedia.org/wiki/Head-mounted_display)

closed 2023-03-27 by GustavoSilvera