HARPLab/DReyeVR #94: How to get Vehicle location（X,Y,Z） of more than EgoVehicle?

wx983, 2023-03-21:
When generating traffic flow vehicles (they are set into `autopilot`  )，is there any method to record the vehicle location like  EgoVehicle as following?
[image]

GustavoSilvera, 2023-03-21:
All the other vehicles in the scene (when being recorded) are also tracked in the same recording log file. See the [Carla documentation](https://carla.readthedocs.io/en/latest/ref_recorder_binary_file_format/#packet-6-position).

wx983, 2023-03-22:
Sorry, I don't really understand it `C:\Users\11\AppData\Local\CarlaUE4\Saved\test.log`, which is a kind of binary file that can't be read by people. Can you guide me more? Should I do something like `show_recorder_file_info.py` to transfer it `test.log`?

wx983, 2023-03-25:
> 是的，准确地说。有[Carla 文档](https://carla.readthedocs.io/en/latest/adv_recorder/#sample-python-scripts)。

@GustavoSilvera 
Through `manual_control.py` and `start_recording.py -f 111.log -n 50` and `show_recorder_file_info.py -a -f C:\Users\11\AppData\Local\CarlaUE4\Saved\111.log -s 111.txt `, I have obtained 111.txt, which  includes the locations of EgoVehicle and `manual_control` vehicle, but without vehicles generated by `start_recording.py`.
Can you guide me how to acquire the location of vehicles generated by `start_recording.py`?
[image]

GustavoSilvera, 2023-03-25:
Those vehicles should be tracked in the same file, but if they aren't spawned at the beginning of the file they will probably be spawned later on. Try scrolling down to find them. Look for all the `vehicle.*.*` patterns in the recording file.

wx983, 2023-03-26:
> Those vehicles should be tracked in the same file, but if they aren't spawned at the beginning of the file they will probably be spawned later on. Try scrolling down to find them. Look for all the `vehicle.*.*` patterns in the recording file.

Thanks, I love you!

closed 2023-03-27 by GustavoSilvera

FBanani, 2023-07-06:
@GustavoSilvera 
Hi, I have another question is there any way to measure the miles or km or in other unit that DReyeVR is drived? as well as the location. Because in a curve road it is not practical to get the location!
Thanks