HARPLab/DReyeVR #95: VR and steering wheel issues (VR: Oculus Quest 2, wheel: Formular Force GP)

ande269f, 2023-03-23:
Hi Gustavo
we're having issues when running the DReyeVR program from any possible place (editor, package).
Either the wheel works, or the VR works, but never together.

Software: 
Windows 10 64x
carla 0.9.13 
SteamVR 1.25.6
and the required softare from the guide is all set with the instructed version.

hardware; 
Gforce GTX 2080 Ti
ryzen 5 5600x
Oculus Quest 2
wheel: Formular Force GP

when running in the editor, with the "launch", and when opening the output log, it shows this: 

[image]

but when opening it in "play", the **VR works fine, but the wheel does not do anything**, and no errors show up on the screen, as it usually does when looking for wheel input. 

[image]

my wheel plugin is downloaded, and i have downloaded the driver and GHUB as well 

[image]

and my SteamVR is active during the whole:

[image]

Do you have a good guess of what is going wrong? 
Thank you for your patience!

GustavoSilvera, 2023-03-23:
So we always run our VR experiments through the "Play->VR Mode" dropdown in editor, and in package/standalone we run it as `CarlaUE4.exe -vr` (with the `-vr` flag). The logitech steering wheel not working during VR mode is interesting and I'm not immediately sure what is wrong. 

Can you make sure you are on the latest version of DReyeVR (0.1.2 at time of writing)

Can you [enable the logitech logging from the Config file](https://github.com/HARPLab/DReyeVR/blob/70a15b5b05b135682cc2daf8ddc4f469e9f9f05c/Config/DReyeVRConfig.ini#L121) to see some prints. 

I'd also suggest you try to run in package mode with `-vr` and enable the package-mode-logging as we describe [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Development.md#1-enable-logging-in-shipping-mode) for further debugging.

Also those BP_Spline errors are benign (just issues with Carla's 0.9.13 Town03 Map) so you can ignore those.

ande269f, 2023-03-27:
Thank you, both wheel and vr works together now! 
it worked updating the DReyeVR. :) 
We used this guide:
[image]

found on the Installation guide to DReyeVR: https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md

closed 2023-03-27 by GustavoSilvera