HARPLab/DReyeVR #96: question and issue about Scenario_runner and DReyeVR

ande269f, 2023-03-27:
Hello 
We have two questions about using scenario_runner with DReyeVR 
we are trying to make our own scenario for studying multimodality in driving. 
But when researching scenario_runner, all the guides that we have found is about using it in carla. And also we encountered an issue while trying out the examples in \srunner

1: (issue) when running these examples: 
[image]

like this: 

[image]

no other vehicles or pedestrians spawns, and the scenarioes do not get activated. There are no errors getting written out in the console:

[image]

2: (question) When making our own simulation environment, how does "routes" and "scenarioes" differ from carla to DReyeVR? can we use the same documentation for understanding it as the scenario_runner documentation made for carla, or should we think about it differently?

GustavoSilvera, 2023-03-27:
Hi, I'll answer your questions in order:
1. There are a few things you should verify before ScenarioRunner should be compatible with your system: You are using Python <=3.8, you are using the scenario-runner `v0.9.13` branch, and that you have installed our DReyeVR patches over if with the `make install` script. This is mentioned inthe DReyeVR [installation guide](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#installing-dreyevr-core) in case you missed it. Can you also try the following sanity check to see if ScenarioRunner is available from your Python environment.
    ```python
    import carla                               # tests carla PythonAPI
    from DReyeVR_utils import find_ego_vehicle # tests DReyeVR PythonAPI
    from scenario_runner import ScenarioRunner # tests scenario-runner
    print(dir(carla))           # to see if carla has everything it needs
    print(dir(ScenarioRunner))  # to see if ScenarioRunner has everything it needs
    ```
2. You should be able to use the vanilla Carla documentation for everything we support (we try to be 100% compatible with the carla features we do support) but we currently only support **routes** in scenario runner. 

We've also taken some action to create our own experiment-runner that should automatically handle the scenario runner and recording. The usage can be explained [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#run-a-scenariorunner-instance-and-record)

ande269f, 2023-03-29:
hello Gustavo. 
thank you so much for your time. 
we have looked at our issue from multiple perspectives now but have still not found a great solution. 

as for your instructions to my question 1: 
everything looks like its good: i have python 3.7.2, sanity check with print(dir(carla)) and print(dir(ScenarioRunner)) went without any trouble, carla and senariorunner seems to be up to date (scenariorunner is v0.9.13), and we updated the DReyeVR recently through the make install script.

but i think the problem is that we dont quite understand DReyeVR and scenario_runner, so if you like, **we would love to ask you some questions that shouldn't take too long to answer:**

So only `--route` is supportet in DReyeVR at the moment, and a route-based scenario command is build as follows:
```
python scenario_runner.py --route <path/to/route-file> <path/to/scenario_sample_file> [route id]
```

that means for making our own scenario and route, we must make 
1. our own **custom_route.xml** 
2. our own **custom_scenario.json**

these files contain either a set of cordinates:

```
 "available_event_configurations": [
                        {
                            "transform": {
                                "pitch": "0",
                                "x": 68.39,
                                "y": 5.52,
                                "yaw": 359.0,
                                "z": 1
                            }
                        },
```

 or a set of waypoints: 
```
<waypoint pitch="366.3597106933594" roll="0.0" x="64.205078125" y="-133.39605712890625" yaw="1.227264404296875" z="7.449135780334473" />
```

so to make sense of that we looked at how to make a route, and checked out traffic-generation-editor for carla (https://github.com/carla-simulator/traffic-generation-editor), but we can't stop woundering why we're the only ones who have asked about this, if it's because we have overlooked something so basic that everyone else but us gets it. 

**question:** 

What is the standard procedure to make routes and scenarios in the `--route` mode? 
Is it by using add-ons like traffic-generation-editor, or by hard-coding it ourselves by sitting in the UE4 carla editor and picking out waypoints?

for scenarios, we looked at the carla dokumentation for creating scenarios (https://carla-scenariorunner.readthedocs.io/en/latest/creating_new_scenario/), but these are made as python .py files, and as we use .json files for the route scenarios, so it's is not the right place to look. Is this true?

We have worked on these questions for the past two days now and would kill for a point in the right direction.

GustavoSilvera, 2023-03-29:
Overall, DReyeVR shouldn't have any effect on the creation process of routes/scenarios using ScenarioRunner, you should be able to use the same scripts you would have previously used on vanilla Carla. Our patches onto scenario-runner (using `make install SR=...` just ensure that the "hero" vehicle is set to the DReyeVR EgoVehicle rather than spawning a new "hero" vehicle actor. 

The standard procedure to creating scenarios is as you mentioned to created a custom route.xml and scenario.json file identified by some integer, and with a few manually placed waypoints (we haven't used the traffic-generation-editor, we just go into the editor to find coordinates in the map). 

The documentation you linked for creating scenarios in Python is specifically for creating "behavior trees" that ScenarioRunner uses to model specific behaviors for the actors to follow. For instance [this](https://github.com/carla-simulator/scenario_runner/blob/v0.9.13/srunner/scenarios/cut_in.py) file encodes the behavior for another vehicle to cut in on the Hero vehicle when certain trigger distances are met. For developing these behaviors yourself it is useful to read into the sequential `py_trees.composites.Sequence` and parallel `py_trees.composites.Parallel` behavior trees (basically whether conditions are blocking or can be run simultaneously). 

I'd recommend trying to run one of Carla's own example scenarios before developing your own from scratch, then you can see how they work and build off those examples. 

As mentioned before, the way to invoke scenario runner with DReyeVR requires the use of the `--route` flag, so running the scenario should look something like this:

```bash
python scenario_runner.py  --route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 10 --output --timeout 1000 --reloadWorld --sync
# where 10 is used to pick which scenario to run from the available ones in the given route files
```

ande269f, 2023-03-30:
Okay, i tried out the editor to find the waypoint coordinates for making a custom route: 

[image]

but:
1. they seem different from the values in the XML files. 
2. The routePlanner marks do not seem to have individual coordinate values 

Is there another way of finding these coordinates in the editor, or should they be transformed in some way before they can be used for making a route?

and as for route scenarios:
 
1. The python scripts, such as the scripts in the %SCENARIO_RUNNER_ROOT%\srunner\scenarios

- ` cut_in.py`
- `change_lane.py`
- `follow_leading_vehicle.py`

are the behavior tree scripts **that define the scenarios**, and the .json file is the file where these scenarios gets **coordinates to where they need to happen**? 

And then these coordinates for the scenario are defined like this in the .json file?

```
                        {
                            "transform": {
                                "pitch": "358.8267822265625",
                                "x": "242.86146545410156",
                                "y": "113.50442504882812",
                                "yaw": "271.3932189941406",
                                "z": "0.5679125189781189"
                            }
                        }, 
```

2. if that is understood correctly, then would this also mean that the 10 scenarios in the `all_towns_traffic_scenarios.json` example: 

[image]

correspond to the listed scenarios in the python script `route_scenario.py`?

```
NUMBER_CLASS_TRANSLATION = {
    "Scenario1": ControlLoss,
    "Scenario2": FollowLeadingVehicle,
    "Scenario3": DynamicObjectCrossing,
    "Scenario4": VehicleTurningRoute,
    "Scenario5": OtherLeadingVehicle,
    "Scenario6": ManeuverOppositeDirection,
    "Scenario7": SignalJunctionCrossingRoute,
    "Scenario8": SignalJunctionCrossingRoute,
    "Scenario9": SignalJunctionCrossingRoute,
    "Scenario10": NoSignalJunctionCrossingRoute
}

```

GustavoSilvera, 2023-03-30:
Yes, I believe there is a slight inconsistency where ScenarioRunner uses centimeters while the UE4Editor uses meters for their primary distance unit... So you'll have to mentally divide the UE4 coordinates by 100 to get them into ScenarioRunner coordinates. 

As for 1.) Yes I believe that is correct. You can think of the .xml as defining the route that the the EgoVehicle takes, and the json file as a list of potential transforms (location/rotation) for where triggers could be placed along this route for particular behaviors. The official Carla documentation is somewhat lacking in this specificity but the [relevant source code](https://github.com/carla-simulator/scenario_runner/blob/v0.9.13/srunner/tools/route_parser.py#L263) is a good resource to read through. 

As for 2.) Yes, that sounds about right to me. But again the [source code](https://github.com/carla-simulator/scenario_runner/blob/f1a94516db57612f22d5c42e597e350e781b9224/srunner/scenarios/route_scenario.py#L325) is a better reference for this than I am, so be sure to double check.

GustavoSilvera, 2023-04-05:
Directional signs are a working feature that should be available to be placed dynamically (ie. not hardcoded/cooked into the map via Editor) at runtime via ScenarioRunner. As per the bottom section of this [documentation](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Signs.md#step-5-optional-registering-with-the-bp-library) you should have the DReyeVR signs listed as static props in the `Default.Package.json` file before running `make package`, this will allow the packaged build to spawn these static actors (props). 

I believe the file you are referring to is in [this dev version](https://github.com/HARPLab/scenario_runner/blob/DReyeVR-0.9.13-dev/srunner/data/all_routes_signs.json) of ScenarioRunner which we haven't updated in a while (sorry!). This is a convenience file that lets us parse and add our custom DReyeVR signs to an entire map similar to other scenario-runner config files. The code logic for this (which you should be able to copy/paste into a vanilla ScenarioRunner repository is [here](https://github.com/HARPLab/scenario_runner/blob/8311dd96922c4c0b5b5e62b7a7c9c071eb21747c/srunner/scenariomanager/scenario_manager.py#L120-L123) to find the file, [here](https://github.com/HARPLab/scenario_runner/blob/8311dd96922c4c0b5b5e62b7a7c9c071eb21747c/srunner/tools/route_parser.py#L47-L66) to read/parse the file, then finally [here](https://github.com/HARPLab/scenario_runner/blob/8311dd96922c4c0b5b5e62b7a7c9c071eb21747c/srunner/scenariomanager/scenario_manager.py#L125-L138) to spawn these signs according to the route. 

We will be improving the documentation to include these tips in the future!

ande269f, 2023-04-06:
We have tried copy/pasting the code from your linked spots into our own scenario_runner 0.9.13 code, but it does not seem to work. We experience these errors: 

this issue when passing the all_signs_scenario.json example 
[image]

this issue when passing our own custom example that is equal to the example i send you in the last message:
[image]

it has something to do with these lines of code in r`oute_parser.py`: 

```
55        with open(signs_filename, 'r') as f:
56            signs_dict = json.loads(f.read())
```
that wants to find the path in `scenario_manager.py` described here: 

```
118        from srunner.tools.route_parser import RouteParser
119        signs_file = os.path.join("..", "data", "all_routes_signs.json")  # scenario_file
120        route_signs_list = RouteParser.parse_direction_signs_file(signs_file, self.single_route)
```
and its not able to find it. 
we are not sure how the code works, but we tried a few things and haven't been able to make it work.
could a solution be just getting the dev version of the scenario_runner code?

GustavoSilvera, 2023-04-06:
Yes, you can try getting the dev version of scenario runner, apologies for it being kinda a mess. It should still fully work just like the vanilla scenario-runner, but with additional patches for DReyeVR specific features.

GustavoSilvera, 2023-04-06:
You can also hardcode the path for the `json.loads()` call to the `all_routes_signs.json` file, then the file should be found and readable.

closed 2023-04-17 by ande269f