HARPLab/DReyeVR #97: CarlaUE4 editor crashes after pressing play

Dreyevr, 2023-03-27:
We tried to use a different map than the default ones from carla using Roadrunner. We followed the steps listed in the carla website and succesfully imported the map into carla (the map injestion steps). To open the map however, we are supposed to open it in the editor and then click on play to use it according to the steps.

But pressing play crashes the editor by displaying fatal error. We did however discover that we can launch DReyeVR by pressing launch instead.

We don't know how this will effect our future experiments. Do you perhaps know where the fault could lie?

We were able to change the default maps using a python script, but it isn't clear how we are supposed to do that with a map exported from roadrunner. Do you maybe know where we could find more info on this?

GustavoSilvera, 2023-03-30:
Okay so according to the logs it looks like not only the SteamVR device is failing to mount but also the SRanipal eye tracking is failing which may cause downstream problems. 

Code-wise the HMD is failing because the call to [`UHeadMountedDisplayFunctionLibrary::IsHeadMountedDisplayEnabled()`](https://docs.unrealengine.com/4.27/en-US/API/Runtime/HeadMountedDisplay/UHeadMountedDisplayFunctionLibra-/IsHeadMountedDis-/) must be failing. I'm not immediately sure why this is the case, you have SteamVR enabled and running alongside DReyeVR (CarlaUE4 Editor)? This might be a benign warning and the HMD might not be initialized just yet (per [this](https://forums.unrealengine.com/t/isheadmounteddisplayenabled-always-returning-false/298120/3)). 

However, the SRanipal error is worrisome and the [status indicator](https://github.com/HARPLab/DReyeVR/blob/70a15b5b05b135682cc2daf8ddc4f469e9f9f05c/DReyeVR/EgoSensor.cpp#L158) (3) indicates that the `Status == SRanipalEye_Framework::FrameworkStatus::ERROR`, suggesting something is awry with your SRanipal installation. For testing purposes can you try disabling SRanipal (in the [build file](https://github.com/HARPLab/DReyeVR/blob/70a15b5b05b135682cc2daf8ddc4f469e9f9f05c/CarlaUE4/CarlaUE4.Build.cs#L46)) and trying again? We can diagnose whether SRanipal is the culprit here or something else is.

Dreyevr, 2023-03-31:
Doing that does indeed 'solve' the issue.

During the installation of SRanipal, we had to use the following solution:
https://www.reddit.com/r/Vive/comments/10ypn77/vive_face_facial_tracker_problems_solved_error/ :

Without this we kept getting errors. Could the problem be here?

When running the packaged version, SRanipal seemed to be working because we had to calibrate it and that went succesfully.

Now after setting the useSRanipal to false and using the editor, we still get a purple indicator that points to where our eyes point to. Does that mean SRanipal is still working then?

GustavoSilvera, 2023-03-31:
Interesting, I have never seen this SRanipal installation issue... there could be something funky going on...

As long as the packaged version works with SRanipal then I imagine this is a bug with SRanipal rather than with DReyeVR/Carla/UE4, so I'm not sure we can do much here. The simplest "solution" is to disable SRanipal for your editor builds (`make launch`) but re-enable it for your shipping/package builds (`make package`) which you will deploy. 

In the editor without SRanipal, the purple debug indicator (wireframe sphere) should just be going in circles, this is the default behavior for when there is no eye tracker attached. If the debug indicator (and screen-space reticle) is actually following your eye gaze then SRanipal is still enabled because otherwise it wouldn't get compiled.

Dreyevr, 2023-04-06:
Do you know how to open a custom map using the packaged version? We only succeeded in changing to default maps. This because PythonAPI probably only looks at a specific directory when a map name is specified. After following the steps to import a map, we get a folder called map_packaged. 

Should we transfer the files to the place where the default maps are? We wanted to try this but we weren't sure which filed go where.

GustavoSilvera, 2023-04-06:
I have actually never tried opening a custom (non Carla default) map in the packaged version, so I may be underqualified to answer this. But I have experienced annoying "cooking" issues with the UE4 package mode and custom assets so I can try to help here:

You should first place your new custom map in the same `Unreal/CarlaUE4/Content/Maps/` directory as all the rest, then you'll probably need to edit [this config file (DefaultGame.ini)](https://github.com/carla-simulator/carla/blob/0.9.13/Unreal/CarlaUE4/Config/DefaultGame.ini#L52-L64) to tell UE4 to "cook" these assets as well. 

Specifically you'll want to point UE4 to "Cook" your custom maps, then you should be able to open it in the package mode either through PythonAPI or by changing the default map in [the Startup settings here (DefaultEngine.ini)](https://github.com/carla-simulator/carla/blob/a1b37f7f1cf34b0f6f77973c469926ea368d1507/Unreal/CarlaUE4/Config/DefaultEngine.ini#L11-L18).

GustavoSilvera, 2023-04-17:
Interesting. There are no worrying error messages in the logfile which suggests a ["FATAL ERROR"](https://forums.unrealengine.com/tags/intersection/unreal-engine/fatal-error) on Windows that is hard to debug. 

But I do notice that there is a (DReyeVR) warning regarding "No SteamVR head mounted device enabled!" which means there might be some connection issue or you are starting the game in flat-screen-mode. 

Unfortunately there is an issue where starting the DReyeVR game in flat-screen mode sometimes causes it to instantly crash like this (no error messages, just dies). But if we then open the game in VR-preview mode, it works fine. We are not sure if this is a bug with UE4, but are looking into it. 

Can you try opening DReyeVR in VR-preview mode? What I've personally found is that every other time I open in VR-preview mode, it doesn't enable the VR anyways so its like "flat-screen-mode" but at least it doesn't crash. 

Also I just want to make sure you can still run the packaged game, is this correct?

Dreyevr, 2023-04-17:
Opening it up in VR-preview mode does seem to do the trick even with SRanipal enabled. We aren't interested in the "flat-screen-mode" anyway. The packaged version works as usual. Thanks for the help.

In an earlier post you told us to put our map in the `Unreal/CarlaUE4/Content/Maps/` directory. But the contents of the that directory aren't exactly like the contents in out custom map. 

After importing the map from roadrunner like suggested in carla/readthedocs, we have a map that contains 3 folder:
**Static**, **Maps** and **config**. **Maps** contains a NAV, OpenDrive and TM folder along with uassets files and UMAP file. **Config** contains 2 JSON files and **Static** more uassets files.

 The `Unreal/CarlaUE4/Content/Maps/` directory only seems to contain one uasset file per UMAP file(town). They do however have a town11 folder that looks similar to our **Maps** folder, however it contains more than one UMAP file and also isn't accessible to us using the following script:
```
import carla

# connect to the CARLA server
client = carla.Client('localhost', 2000)
client.set_timeout(10.0)

# get the world object
world = client.get_world()

# load a new map
new_map = 'Town11'
world = client.load_world(new_map)
```
We are only able to change to maps 1-7 and 10. Simply moving our whole map folder to `Unreal/CarlaUE4/Content/Maps/` wouldn't work in this case since we don't know which path the above script looks in for the new_map. I hope it is clear what I'm trying to say.

Opening the map in the editor does work however. We are just not sure how to open it using the packaged version using a script similar to the one above.

GustavoSilvera, 2023-04-17:
Have you added your new map into the [`+MapsToCook`](https://github.com/HARPLab/DReyeVR/blob/70a15b5b05b135682cc2daf8ddc4f469e9f9f05c/Config/DefaultGame.ini#L52-L69) configuration? This should allow you to cook your new maps and access them from the packaged game.

closed 2023-05-10 by GustavoSilvera