HARPLab/DReyeVR #98: Just bicycle made

elnazbanani, 2023-03-31:
Hi. Although I call the vehicle with attribute ('number_of_wheels')) == 4, it just provide bicycle. this is my code. could you help me?

import time
import carla
#import numpy as np
#from DReyeVR_utils import find_ego_vehicle

from carla import VehicleLightState as vls

import argparse
import logging

from numpy import random

#---------------get actor----------------------------
def get_actor_blueprints(world, filter, generation):
    bps = world.get_blueprint_library().filter(filter)

    if generation.lower() == "all":
        return bps

    # If the filter returns only one bp, we assume that this one needed
    # and therefore, we ignore the generation
    if len(bps) == 1:
        return bps

    try:
        int_generation = int(generation)
        # Check if generation is in available generations
        if int_generation in [1, 2]:
            bps = [x for x in bps if int(x.get_attribute('generation')) == int_generation]
            return bps
        else:
            print("   Warning! Actor Generation is not valid. No actor will be spawned.")
            return []
    except:
        print("   Warning! Actor Generation is not valid. No actor will be spawned.")
        return []
#---------------------------------------------------------------
def main():
    argparser = argparse.ArgumentParser(
        description=__doc__)
    argparser.add_argument(
        '--host',
        metavar='H',
        default='127.0.0.1',
        help='IP of the host server (default: 127.0.0.1)')
    argparser.add_argument(
        '-p', '--port',
        metavar='P',
        default=2000,
        type=int,
        help='TCP port to listen to (default: 2000)')
    argparser.add_argument(
        '-n', '--number-of-vehicles',
        metavar='N',
        default=50,
        type=int,
        help='Number of vehicles (default: 30)')
    argparser.add_argument(
        '--safe',
        action='store_true',
        help='Avoid spawning vehicles prone to accidents')
    argparser.add_argument(
        '--filterv',
        metavar='PATTERN',
        default='vehicle.*',
        help='Filter vehicle model (default: "vehicle.*")')
    argparser.add_argument(
        '--generationv',
[... 209 lines cut ...]
                elif counter == 0:
                    counter = spawn_delay
            else:
                world.wait_for_tick()

    finally:
        
        if not args.asynch and synchronous_master:
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.no_rendering_mode = False
            settings.fixed_delta_seconds = None
            world.apply_settings(settings)
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])

        	

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')

GustavoSilvera, 2023-03-31:
Hi there, are you passing the `--safe` flag in to enable `args.safe` that would enable your filter that `number_of_wheels == 4`?

elnazbanani, 2023-04-01:
thank you, I have another issue please help me:
I search a lot and I understand try_spawn_actor could be None but how can I fixt it. my spawn point has z=0.
[image]
as you can see, I have vehicle_bp and also spawn point but I could noy understand why I use try I got vehicle=None 
[image]
how can I fix it?

GustavoSilvera, 2023-04-02:
Try setting the height a little higher, like z=0.1. The actor might be spawning inside the ground or your ground may not be at 0 because the map is not necessarily level everywhere. You should instead look into carla's recommended spawn-point locations to spawn your actors. 

Also you should look through [Carla's PythonAPI examples](https://github.com/carla-simulator/carla/tree/master/PythonAPI/examples) to see how they do it.

elnazbanani, 2023-04-03:
Hi I have issues, Could you help me?
1) I want to spawn a vehicle in 4 different lanes with consideration of each route. but when I run the following code it just provided vehicles in route 1 and 3. what should I do to fix it?
2) Also, the type of vehicles was the same in each lane and did not change.
3) Although, I do not allow the vehicle to change the lanes, they changed their lane.

[image]

---------------------------------------code--------------------------------
import time
import carla
#import numpy as np
#from DReyeVR_utils import find_ego_vehicle
#------------------------------------------------------ 
##with VehicleLightState we can manage the position of the light in the carla:
#Class that recaps the state of the lights of a vehicle, these can be used as a flags.
#E.g: VehicleLightState.HighBeam & VehicleLightState.Brake will return True when both are active.
#Lights are off by default in any situation and should be managed by the user via script. 
#The blinkers blink automatically. Warning: Right now, not all vehicles have been prepared to
#work with this functionality, this will be added to all of them in later updates.
#------------------------------------------------------
from carla import VehicleLightState as vls

import argparse
import logging
#--------------------------------------------------------
#Logs provide developers with an extra set of eyes that are constantly looking at the flow 
#that an application is going through. They can store information, like which user or IP accessed
#the application. If an error occurs, then they can provide more insights than a stack trace by
#telling you what the state of the program was before it arrived at the line of code where the 
#error occurred.
#By logging useful data from the right places, you can not only debug errors easily but also use
#the data to analyze the performance of the application to plan for scaling or look at usage 
#patterns to plan for marketing.
#levels:debug, info, warning, error, critical
#logging.debug('This is a debug message')
#logging.info('This is an info message')
#logging.warning('This is a warning message')
#logging.error('This is an error message')
#logging.critical('This is a critical message')
#---------------------------------------------------------
from numpy import random

#---------------get actor----------------------------
def get_actor_blueprints(world, filter, generation):
    bps = world.get_blueprint_library().filter(filter)

    if generation.lower() == "all":
        return bps

    # If the filter returns only one bp, we assume that this one needed
    # and therefore, we ignore the generation
    if len(bps) == 1:
        return bps

    try:
        int_generation = int(generation)
        # Check if generation is in available generations
        if int_generation in [1, 2]:
            bps = [x for x in bps if int(x.get_attribute('generation')) == int_generation]
            return bps
        else:
            print("   Warning! Actor Generation is not valid. No actor will be spawned.")
            return []
    except:
        print("   Warning! Actor Generation is not valid. No actor will be spawned.")
        return []
#---------------------------------------------------------------
def main():
    argparser = argparse.ArgumentParser(
        description=__doc__)
[... 293 lines cut ...]
                world.wait_for_tick()

    finally:
        
        if not args.asynch and synchronous_master:
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.no_rendering_mode = False
            settings.fixed_delta_seconds = None
            world.apply_settings(settings)
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list1])
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list2])
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list3])
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list4])

        	

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')

elnazbanani, 2023-04-03:
> Try setting the height a little higher, like z=0.1. The actor might be spawning inside the ground or your ground may not be at 0 because the map is not necessarily level everywhere. You should [...]

closed 2023-04-03 by elnazbanani

elnazbanani, 2023-04-03:
Hi, I want to generate vehicles in 4 different lanes but I have some issues as you can see in the following picture:
1) vehicles just provided in lane 1 and 3.
2)all of the vehicles are the same.
3)Although I did not allow them to change their lane, they changed it.
Could you help me?
[image]
---------------------code------------------------
import time
import carla
from carla import VehicleLightState as vls
import argparse
import logging
from numpy import random
def get_actor_blueprints(world, filter, generation):
    bps = world.get_blueprint_library().filter(filter)

    if generation.lower() == "all":
        return bps

    # If the filter returns only one bp, we assume that this one needed
    # and therefore, we ignore the generation
    if len(bps) == 1:
        return bps

    try:
        int_generation = int(generation)
        # Check if generation is in available generations
        if int_generation in [1, 2]:
            bps = [x for x in bps if int(x.get_attribute('generation')) == int_generation]
            return bps
        else:
            print("   Warning! Actor Generation is not valid. No actor will be spawned.")
            return []
    except:
        print("   Warning! Actor Generation is not valid. No actor will be spawned.")
        return []
#---------------------------------------------------------------
def main():
    argparser = argparse.ArgumentParser(
        description=__doc__)
    argparser.add_argument(
        '--host',
        metavar='H',
        default='127.0.0.1',
        help='IP of the host server (default: 127.0.0.1)')
    argparser.add_argument(
        '-p', '--port',
        metavar='P',
        default=2000,
        type=int,
        help='TCP port to listen to (default: 2000)')
    argparser.add_argument(
        '-n', '--number-of-vehicles',
        metavar='N',
        default=50,
        type=int,
        help='Number of vehicles (default: 30)')
    argparser.add_argument(
        '--safe',
        action='store_true',
        default= True,
        help='Avoid spawning vehicles prone to accidents')
    argparser.add_argument(
        '--filterv',
        metavar='PATTERN',
        default='vehicle.*',
        help='Filter vehicle model (default: "vehicle.*")')
    argparser.add_argument(
        '--generationv',
        metavar='G',
[... 247 lines cut ...]
                            alt = True
                            alt2 = True

                    

                    counter -= 1

                elif counter > 0:
                    counter -= 1
                elif counter == 0:
                    counter = spawn_delay
            else:
                world.wait_for_tick()
    finally:
        
        if not args.asynch and synchronous_master:
            settings = world.get_settings()
            settings.synchronous_mode = False
            settings.no_rendering_mode = False
            settings.fixed_delta_seconds = None
            world.apply_settings(settings)
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list1])
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list2])
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list3])
            client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list4])