HARPLab/DReyeVR #101: 'Actor' object has no attribute 'set_autopilot'

whyhowie, 2023-04-11:
Greetings, 

I was trying to run the python scripts in the example folder (e.g. `DreyeVR_AI.py`). However, it looks like the `set_autopilot` attribute was missing for the vehicle created. Any possible reasons for this issue? Thanks!

(Currently I'm running the server without a VR headset and a steering wheel connected. Could this be related?)

Traceback (most recent call last):
  File "DReyeVR_AI.py", line 152, in <module>
    main()
  File "DReyeVR_AI.py", line 131, in main
    ego_vehicle = set_DReyeVR_autopilot(world, traffic_manager)
  File "DReyeVR_AI.py", line 32, in set_DReyeVR_autopilot
    DReyeVR_vehicle.set_autopilot(True, traffic_manager.get_port())
AttributeError: 'Actor' object has no attribute 'set_autopilot'

GustavoSilvera, 2023-04-11:
Interesting. You shouldn't need the VR headset nor steering wheel for this functionality to work. 

According to the [documentation](https://carla.readthedocs.io/en/0.9.13/python_api/) all vehicles should have a `set_autopilot`, so what I believe is happening is that the `find_ego_vehicle` is returning a `carla.libcarla.Actor` rather than a `carla.libcarla.Vehicle` class. 

Can you verify you are on the latest version of DReyeVR (0.2) and print the `DReyeVR_vehicle` after obtaining this reference from `find_ego_vehicle()`?

Also, just as a sanity check, did you rebuild the PythonAPI after installing DReyeVR to Carla?

whyhowie, 2023-04-11:
Thanks for your reply. Yes, it's quite odd... I just did `make clean`, and then `make PythonAPI && make launch`. However, I think the type of the `ego_vehicle` found is still `Actor`.

Here's what I printed out for the vehicle in `set_DReyeVR_autopilot()`:
> Actor(id=86, type=harplab.dreyevr_vehicle.model3)
> <class 'carla.libcarla.Actor'>

I'm currently running DReyeVR 0.1.2 and carla 0.9.13. The Unreal Engine server is up and running despite missing the steering wheel and the vr headset.
[image]

Thanks for your help!

whyhowie, 2023-04-14:
Hi Gustavo,

Thanks! I `--force-reinstall` 'ed the `.whl` file and it seems that the error is gone. I might have never used the force reinstall option before. Hopefully the following output looks reasonable:
[image]

However, not sure if it is relevant but I get a different error in `schematic_mode.py`, which involves `number_of_wheels`:
[image]

GustavoSilvera, 2023-04-14:
Yup, this looks correct now. (you can delete the `std::cout` prints and rebuild the PythonAPI if you want). 

The `number_of_wheels` bug is a known issue that had been discussed here #100