HARPLab/DReyeVR #102: How to set the specific bitrh location,  moving path for vehicle using  python script？

wx983, 2023-04-11:
Hi! I want to generate one autonomous vehicle and I wish to set a specific moving path for it.
I know how to generate one self-driving vehicle but I don't know how to create a specific moving path for it.
It's about a question about Carla not DReyeVR. Do you know how to do that?

GustavoSilvera, 2023-04-11:
This is definitely a Carla question rather than a DReyeVR question, and I'd recommend taking a look at this [Carla issue](https://github.com/carla-simulator/carla/issues/4286) which seems to address it.

wx983, 2023-04-11:
> This is definitely a Carla question rather than a DReyeVR question, and I'd recommend taking a look at this [Carla issue](https://github.com/carla-simulator/carla/issues/4286) which seems to address it.

According to what you say, I found it. It seems to be able to address my problem.
[image]

But what should I do next? The following codes are to generate cars and set them to autonomous mode. But I don't know how to acquire `actor` above in red rectangular part for autonomous vehicle.

`batch = []
        for n, transform in enumerate(spawn_points):
            if n >= count:
                break
            
            blueprint = world.get_blueprint_library().find('vehicle.lincoln.mkz_2020')
            # blueprint = world.get_blueprint_library().find('vehicle.audi.a2')
            assert blueprint is not None
            blueprint.set_attribute('role_name', 'autopilot')
            batch.append(SpawnActor(blueprint, transform).then(SetAutopilot(FutureActor, True)))`

And as for `path` in rectangular part, 

`from carla import RoadOption
route = start_waypoint.plan_route_to_waypoint (end_waypoint,[])
route_waypoints =[]
for wp in route:
    route_waypoints.append (wp.transform.location)
`
I think `route_waypoints` can be used as `path`.

GustavoSilvera, 2023-04-12:
I'm not 100% sure, but I believe once you spawn the actors using the `batch.append(SpawnActor(...))` code you have, then you'l have to apply the batch command like this:
```python
vehicles_list = []
for response in client.apply_batch_sync(batch, synchronous_master):
    if response.error:
        logging.error(response.error)
    else:
        vehicles_list.append(response.actor_id)
```

Then you have a list of all the vehicles (in `vehicles_list`) which you can iterate through as follows:
```python
all_vehicle_actors = world.get_actors(vehicles_list)
for actor in all_vehicle_actors:
    traffic_manager.update_vehicle_lights(actor, True)
```

You should refer to this [example script](https://github.com/carla-simulator/carla/blob/f14acb257ebf44c302b225b02080ac5f0eedcf7f/PythonAPI/examples/generate_traffic.py#L240-L253) on how Carla does this.

closed 2023-05-10 by GustavoSilvera

Nitro60zeus, 2023-08-26:
> > This is definitely a Carla question rather than a DReyeVR question, and I'd recommend taking a look at this [Carla issue](https://github.com/carla-simulator/carla/issues/4286) which seems to address it.
> 
> According to what you say, I found it. It seems to be able to address my problem. [image]
> 
> But what should I do next? The following codes are to generate cars and set them to autonomous mode. But I don't know how to acquire `actor` above in red rectangular part for autonomous vehicle.
> 
> ```
> batch = []
> for n, transform in enumerate(spawn_points):
>     if n >= count:
>         break
>     
>     blueprint = world.get_blueprint_library().find('vehicle.lincoln.mkz_2020')
>     # blueprint = world.get_blueprint_library().find('vehicle.audi.a2')
>     assert blueprint is not None
>     blueprint.set_attribute('role_name', 'autopilot')
>     batch.append(SpawnActor(blueprint, transform).then(SetAutopilot(FutureActor, True)))
> ```
> 
> And as for `path` in red rectangular part,
> 
> ```
> from carla import RoadOption
> route = start_waypoint.plan_route_to_waypoint (end_waypoint,[])
> route_waypoints =[]
> for wp in route:
>     route_waypoints.append (wp.transform.location)
> ```
> 
> I think `route_waypoints` can be used as `path`.

Hi! Were you finally able to run custom path on autopilot?

wx983, 2023-09-15:
> Hi! Were you finally able to run custom path on autopilot?

Sorry, I still can't. I can only use sumo-carla cosimulation to control the specific path for vehicles, but not on autopilot.

Nitro60zeus, 2023-09-15:
Oh. I figured it out though. There's two ways you can do.

1. Using npc_agent (uses scenario manager): 
After preparing scenario, you just put the agent to drive on the designed route. 
`py scenario_runner.py --route "path to route xml file" "path to traffic scenario json file" 0 --sync --agent "path to agent( --/srunner/autoagents/npc_agent.py)"` 

2. Using waypoints (uses traffic manager):
Get the spawn points on the route you want to decide and put them in ` automatic_control.py` using this function: `traffic_manager.set_path(vehicle, route)`. 

Lemme know if u need additional details.

wx983, 2023-10-24:
> Oh. I figured it out though. There's two ways you can do.
>
> 1. Using npc_agent (uses scenario manager):
>    After preparing scenario, you just put the agent to drive on the designed route.
>    `py [...]

Hi! About the second way, could you please give me one successful example including all codes and commands? I wonder how to get the spawn points in Town03 Map built-in CARLA and where to insert `traffic_manager.set_path(vehicle, route)` in `automatic_control.py`.

wx983, 2023-10-24:
@Nitro60zeus And I also wonder if those two ways you mentioned before obey some form of autonomous driving algorithms. 
I mean, for example, `DReyeVR_vehicle.set_autopilot(True, traffic_manager.get_port())` obey the autonomous driving algorithm called `autopilot`, which can be seen [here](https://carla.readthedocs.io/en/0.9.13/adv_traffic_manager/#in-memory-map:~:text=The%20control%20loop%20consists%20of%20five%20different%20stages)
Thank you!

Nitro60zeus, 2023-11-20:
in DReyeVR_AI.py, add the following changes:

```
def set_DReyeVR_autopilot(world, traffic_manager):
    DReyeVR_vehicle = find_ego_vehicle(world)
    if DReyeVR_vehicle is not None:
        DReyeVR_vehicle.set_autopilot(True, traffic_manager.get_port())
        

        scenario_xml_path = "C:\scenario_runner\srunner\data\exp.xml"
        tree = ET.parse(scenario_xml_path)
        root = tree.getroot()
        
        waypoints = []
        
        for waypoint_elem in root.findall(".//waypoint"):
         x = float(waypoint_elem.get("x"))
         y = float(waypoint_elem.get("y"))
         z = float(waypoint_elem.get("z"))
         
         print(x)
       
         waypoint = carla.Location(x, y, z)
         waypoints.append(waypoint)
        
        print(waypoints)
        
        traffic_manager.set_path(DReyeVR_vehicle,waypoints)
       
        spawn_points = world.get_map().get_spawn_points()

        # Route 
        spawn_point_1 =  spawn_points[32]
        # Create route 1 from the chosen spawn points
        route_1_indices = [222, 164]
        route_1 = []
        for ind in route_1_indices:
         route_1.append(spawn_points[ind].location)
        
        traffic_manager.set_path(DReyeVR_vehicle, route_1)
   
        
        print("Successfully set autopilot on ego vehicle")
        
    return DReyeVR_vehicle
```