HARPLab/DReyeVR #105: scenario_runner error

Dreyevr, 2023-04-20:
We are getting the following error when trying to testing out the demo examples:
```
(carla13) C:\CarlaDReyeVR\scenario_runner>python run_experiment.py --title "dreyevr_experiment" --route %SCENARIO_RUNNER_ROOT%\srunner\data\routes_training.xml %SCENARIO_RUNNER_ROOT%\srunner\data\town05_scenarios.json 0
Starting scenario runner
Recording on file: C:/Users/Administrator/AppData/Local/CarlaUE4/Saved/exp_dreyevr_experiment_04_20_2023_17_41_25.rec
INFO:  Found the required file in cache!  Carla/Maps/TM/Town01.bin
Preparing scenario: RouteScenario_0
The scenario cannot be loaded
Traceback (most recent call last):
  File "C:\CarlaDReyeVR\scenario_runner\scenario_runner.py", line 389, in _load_and_run_scenario
    debug_mode=self._args.debug)
  File "C:\CarlaDReyeVR\scenario_runner\srunner\scenarios\route_scenario.py", line 157, in __init__
    self._update_route(world, config, debug_mode)
  File "C:\CarlaDReyeVR\scenario_runner\srunner\scenarios\route_scenario.py", line 186, in _update_route
    world_annotations = RouteParser.parse_annotations_file(config.scenario_file)
  File "C:\CarlaDReyeVR\scenario_runner\srunner\tools\route_parser.py", line 37, in parse_annotations_file
    with open(annotation_filename, 'r', encoding='utf-8') as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\CarlaDReyeVR\\scenario_runner\\srunner\\data\\town05_scenarios.json'
[Errno 2] No such file or directory: 'C:\\CarlaDReyeVR\\scenario_runner\\srunner\\data\\town05_scenarios.json'
Exception ignored in: <function RouteScenario.__del__ at 0x0000027DDBD74400>
Traceback (most recent call last):
  File "C:\CarlaDReyeVR\scenario_runner\srunner\scenarios\route_scenario.py", line 537, in __del__
    self.remove_all_actors()
  File "C:\CarlaDReyeVR\scenario_runner\srunner\scenarios\basic_scenario.py", line 202, in remove_all_actors
    for i, _ in enumerate(self.other_actors):
AttributeError: 'RouteScenario' object has no attribute 'other_actors'
No more scenarios .... Exiting
Stopped scenario runner, Result: False
Stopping recording, file saved to C:/Users/Administrator/AppData/Local/CarlaUE4/Saved/exp_dreyevr_experiment_04_20_2023_17_41_25.rec
```
Also in the demo examples, the command specifies routes_custom.xml which doesn't seem to exist : https://github.com/carla-simulator/scenario_runner/tree/v0.9.13/srunner/data

I would also like to ask if it is possible to use scenarios made in roadrunner, and if so with the following behaviour: After passing a specific point, a car starts tailgating the driver.

GustavoSilvera, 2023-04-20:
First off, can you try using [our fork](https://github.com/harplab/scenario_runner/commits/DReyeVR-0.9.13) (on branch `DReyeVR-0.9.13`) of ScenarioRunner (which should have the minimal required patches for DReyeVR).

Then make sure to run scenarios with the `--route ROUTE.xml SCENARIOS.json SCENARIO_ID` where `ROUTE` is the path to your route file (ex. srunner/data/routes_training.xml), SCENARIOS is the path to your scenario file (ex. srunner/data/all_towns_traffic_scenarios.json) and ROUTE is an integer denoting which scenario to run (ex. 10)

GustavoSilvera, 2023-04-20:
Looking into this more, and I believe the main reason for this issue is because the `find_ego_vehicle` DReyeVR method is not found. Can you add something like this to right before the `from DReyeVR_utils import find_ego_vehicle`:

```python
import sys, os
sys.path.insert(0, os.path.join(os.getenv("CARLA_ROOT"), "PythonAPI")) # add Carla files
sys.path.insert(0, os.path.join(os.getenv("CARLA_ROOT"), "PythonAPI", "examples")) # add Carla examples
```

Since currently (as of writing) the `DReyeVR_utils.py` file lives in `PythonAPI/examples` (this will be moved to its own directory soon).

Dreyevr, 2023-04-24:
I deleted my previous scenario_runner folder and cloned the one you suggested. So now I am working on the `DReyeVR-0.9.13` branch. When i went to install (make install SR=...) scenario_runner again, I couldn't pass the sanity check.
```
>>> from DReyeVR_utils import find_ego_vehicle
>>> from scenario_runner import ScenarioRunner
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\CarlaDReyeVR\scenario_runner\scenario_runner.py", line 33, in <module>
    sys.path.insert(0, os.path.join(os.getenv("CARLA_ROOT"), "PythonAPI")) # for Carla stuff
  File "C:\Users\Administrator\.conda\envs\carla13\lib\ntpath.py", line 76, in join
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
>>> exit()

(carla13) C:\CarlaDReyeVR>make check
make: *** No rule to make target `check'.  Stop.

(carla13) C:\CarlaDReyeVR>cd DReyeVR

(carla13) C:\CarlaDReyeVR\DReyeVR>make check
python Scripts/check_install.py --carla  --scenario-runner  --verbose
Simple script to check correspondences are present

Unable to find installation or user input for CARLA_ROOT

Unable to find installation or user input for SCENARIO_RUNNER_ROOT
**************************************************

**************************************************
Summary:

Done check
**************************************************
```
This may have to do with my system not recognizing CARLA_ROOT and SCENARIO_RUNNER_ROOT, because when i enter `make check CARLA=../carla SR=../scenario_runner`, I get the expected output.

I made sure to include the path file in the following directories:
C:\Users\Administrator\.conda\envs\carla13\Lib\site-packages
C:\Anaconda3\envs\carla\Lib\site-packages
C:\Anaconda3\Lib\site-packages

GustavoSilvera, 2023-04-24:
(It is also a good idea to keep a `SCENARIO_RUNNER_ROOT` environment variable as well.

Dreyevr, 2023-04-27:
Do the predefined carla scenarios work with DReyeVR? I tried to run the FollowLeadingVehicle_1 scenario as explained here https://carla-scenariorunner.readthedocs.io/en/latest/getting_scenariorunner/ , but nothing seemed to happen.

I thought it was because carla designed the scenarios for autonomous vehicles to follow, but in the link I referenced they talk about testing it out with manual_control.py.

If the scenario is supposed work, is the leading vehicle supposed to appear in front of you on a given position? Or does it appear as soon as you start driving?

GustavoSilvera, 2023-04-27:
Scenario runner should work as long as you are using our patches because we override the logic to [spawn a DReyeVR EgoVehicle](https://github.com/HARPLab/scenario_runner/blob/DReyeVR-0.9.13/srunner/scenarios/route_scenario.py#L172) as the `"hero"` vehicle rather than a normal Carla one. We have only officially tested on scenarios with the `--route` flag however, so you should take a look at those scenarios as discussed here: #96

Dreyevr, 2023-04-27:
I went through the issue as you suggested. So is it not possible to run scenarios with an OpenSCENARIO format? I didn't quite get the answer from the discussion.

GustavoSilvera, 2023-04-27:
Sorry, I specifically wanted to reference [this comment](https://github.com/HARPLab/DReyeVR/issues/96#issuecomment-1489002872) to get ScenarioRunner running with DReyeVR. Namely that you need to run the `scenario_runner.py` script like this:
```sh
python scenario_runner.py  --route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 10 --output --timeout 1000 --reloadWorld --sync
```

Dreyevr, 2023-04-28:
So if we were to export the scenarios in OpenScenario as an xml file + a json file, we'll be able to run our scenarios in the way you described?

GustavoSilvera, 2023-04-28:
If you can run it in `--route` mode, then I would assume so. But this is untested territory for us.

Dreyevr, 2023-05-03:
I tried to run a scenario using the route method to get a better feel for what they are supposed to do in DReyeVR.
I ran the following scenario:
```
python scenario_runner.py  --route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 1 --output --timeout 1000 --reloadWorld --sync
```
This is supposed to be a control_loss scenario. This loads a different town with multiple actors, but I didn't experience any control loss. The scenario seemed to have run succesfully, because it gives a report afterwards. Only I can't seem to understand what it's supposed to do.

Do I only loss control when I follow a specific path?

Also from the issue you as referenced. Therein you say that xml is the route the ego_vehicle has to follow. And the JSON file is the list of potential locations where trigger are placed. Where is then the behaviour of the other actors described? Is that in the python scripts with the behaviour trees? 

If so, then only having the xml file and the JSON file isn't enough. So even if we are able to get an xml file and JSON file from OpenScenario, the actual behaviour will still need to be made in a seperate python file? Which makes the use of OpenScenario in this situation pointless.

GustavoSilvera, 2023-05-03:
The xml and json files (as you mentioned) describe essentially which route/scenario to load, but the corresponding logic for each of these scenarios is (in python) in the [`scenario_runner/srunner/scenarios`](https://github.com/HARPLab/scenario_runner/tree/DReyeVR-0.9.13/srunner/scenarios) directory. For your control-loss scenario ([here](https://github.com/HARPLab/scenario_runner/blob/DReyeVR-0.9.13/srunner/scenarios/control_loss.py)) it looks to me like there is some code to trigger a control-jitter for the EgoVehicle when a certain distance threshold is met. This might not work with DReyeVR since the controls are handled in C++ through the server logitech wheel, so maybe try another scenario such as [`signalized_junction_left_turn`](https://github.com/HARPLab/scenario_runner/blob/DReyeVR-0.9.13/srunner/scenarios/signalized_junction_left_turn.py)

GustavoSilvera, 2023-05-10:
As an improvement, I've been working on a new system to automatically spawn the navigational signs according to the route waypoints. You can check this out in #112

closed 2023-05-10 by GustavoSilvera