HARPLab/DReyeVR #107: Certain controls not available after exiting spectator mode

whyhowie, 2023-04-25:
I'm testing the control handover, and it seems that after returning from spectator mode to manual control (by pressing `1`) or autonomous driving, only the `WASD` keys are working. Other functions such as the turn signals and the reverse gear are no longer available by pressing the corresponding keys. The `Q` and the `E` keys are now bound to camera movement. Could this be a bug or there are some settings that I might have missed?

Thanks!

GustavoSilvera, 2023-04-25:
Just so I can reproduce this bug, can you list out your process in detail:
(this is what I think you are doing, correct me if needed)
1. Open game (in VR mode), can drive around with WASD and wheel hardware (logi)
2. Enter DReyeVR "spectator mode" by pressing `2` and can fly around with WASD,Q,E
3. Repossess the EgoVehicle (leaving spectator mode) by pressing `1`
4. Now the EgoVehicle hardware controls (logi inputs) are broken and only the WASD keys seem to work, the Q,E buttons still raise and lower the camera.

I'll look into this since this sounds like a bug to me.

GustavoSilvera, 2023-04-25:
Interesting, ok. Can you try switching to the `dev` branch and trying this out? We reworked some of the underlying EgoVehicle controls logic and I was unable to reproduce your issue on this branch but I was on the old (current public) branch. Unfortunately since its a fairly large change you'll need to perform a `make clean` before installing the update. Please check out the [Upgrading `DReyeVR` section](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#upgrading-dreyevr). We should be making a PR for this soon.

closed 2024-08-09 by ajdroid