HARPLab/DReyeVR #108: Data extraction from DReyeVR using Scenario_runner

ande269f, 2023-04-25:
Hello 
we're using scenario runner and tried extracting our data by using this program: 

`C:\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples\DReyeVR_logging.py`

it works well when running it without scenario_runner, but gives an error when using  scenario_runner:
```
(carla13) C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples>python DreyeVR_logging.py
Rospy not initialized. Unable to use ROS for logging
No EgoVehicle found, spawning one: "harplab.dreyevr_vehicle.model3"
initialized DReyeVRSensor PythonAPI client

done.
Traceback (most recent call last):
  File "DreyeVR_logging.py", line 154, in <module>
    main()
  File "DreyeVR_logging.py", line 133, in main
    sensor.ego_sensor.listen(publish_and_print)
AttributeError: 'NoneType' object has no attribute 'listen'
```
we're looking at using metrics instead, because we thought that we might have misunderstood the way to get game data. Should we look at metrics, or is DReyeVR designed so that the pythonAPI examples are meant to do the job?

GustavoSilvera, 2023-04-25:
Hi, it looks like this is a DReyeVR bug that is happening because the EgoVehicle is not present in the map, so there is some logic in `find_ego_sensor` that should look for the EgoSensor or spawn the EgoVehicle if it doesn't exist. In your case it looks like it is successfully being spawned but then the EgoSensor is not present (because we don't search through the world actors again!)

Can you try the following:

Replace the `class DReyeVRSensor` initialization method to be:
```python
def __init__(self, world: carla.libcarla.World):
        self.ego_sensor: carla.sensor.dreyevrsensor = find_ego_sensor(world)
        if self.ego_sensor is None: # try again
            self.ego_sensor = find_ego_sensor(world)
        assert self.ego_sensor is not None # sanity check
        self.data: Dict[str, Any] = {}
        print("initialized DReyeVRSensor PythonAPI client")
```

I think this logging was previously working because the EgoVehicle (and thus EgoSensor) was present in the world and didn't need to be spawned, which may not be the case for scenario-runner. 

I think the best way to get these user inputs is via the EgoSensor (we haven't touched the SR metrics) so I think you are doing this as intended.

ande269f, 2023-04-25:
we have noticed that a value for the driver's speed (the speed that is also shown on the "dashboard" inside the DReyeVR vehicle) is not one of the variables in sensor.data. Is there a smart way to get that value as well?

GustavoSilvera, 2023-04-25:
Getting the exact same value from the server to the PythonAPI might be a bit tricky since we'd need to pipe the data through to the PythonAPI (which is somewhat tedious, steps can be found [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Development.md#adding-custom-data-to-the-ego-sensor)). But another method of getting the forward velocity can be through [Carla's own API](https://carla.readthedocs.io/en/latest/python_api/#carlaactor) (You'll probably want to look at `actor.get_velocity()`)

GustavoSilvera, 2023-04-25:
I also noticed regarding my earlier workaround that sometimes the EgoVehicle might fail to spawn before the second check for the EgoSensor is present... which would cause it to try to spawn another EgoVehicle and fail disastrously. This is a race condition and I'm working on a better fix asap.

ande269f, 2023-04-25:
Awesome! We will check it out first thing tomorrow and update you on it

ande269f, 2023-04-26:
i tried the fix for your last comment in DReye_VR_utils.py, but i get an error here: 

   return list(world.get_actors().filter("harplab.dreyevr_sensor.*"))`

it says that "world" is an unresolved reference
i did exacly as was shown in the comment you linked to

when i run my logging in senario runner, i get this: 
```
(carla13) C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples>DReyeVR_logging_vores.py
Traceback (most recent call last):
  File "C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples\DReyeVR_logging_vores.py", line 10, in <module>
    from DReyeVR_utils import DReyeVRSensor
  File "C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples\DReyeVR_utils.py", line 53
    def get_world_sensors() -> list:
      ^
IndentationError: expected an indented block

(carla13) C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples>

```

ande269f, 2023-04-26:
i went back to my previous setup in DReyeVR_utils.py, and i haven's had any problems :) yet

ande269f, 2023-04-26:
Your fix works wounderfully!

We failed on getting out the ego_vehicle speed value for our experiment, i looks very possible, but we lack the skills to do it haha. We use the throttle value from DReyeVR_logging.py instead and then approximate a value from the recordings :)

ande269f, 2023-04-26:
update. We are running into this issue all of a sudden

```
(carla13) C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples>python DReyeVR_logging_vores.py
Rospy not initialized. Unable to use ROS for logging

done.
Traceback (most recent call last):
  File "DReyeVR_logging_vores.py", line 153, in <module>
    main()
  File "DReyeVR_logging_vores.py", line 119, in main
    sensor = DReyeVRSensor(world)
  File "C:\Users\Bruger\carla\Build\UE4Carla\0.9.13-dirty\WindowsNoEditor\PythonAPI\examples\DReyeVR_utils.py", line 73, in __init__
    assert self.ego_sensor is not None  # sanity check
AssertionError
```
i dont think i have changed anything after following your instructions.

this is my code: 
```

import numpy as np
from typing import Any, Dict, List, Optional

import sys
import glob
import os
import time

try:
    sys.path.append(
        glob.glob(
            "../carla/dist/carla-*%d.%d-%s.egg"
            % (
                sys.version_info.major,
                sys.version_info.minor,
                "win-amd64" if os.name == "nt" else "linux-x86_64",
            )
        )[0]
    )
except IndexError:
    pass

import carla

def find_ego_vehicle(world: carla.libcarla.World) -> Optional[carla.libcarla.Vehicle]:
    DReyeVR_vehicle = None
    ego_vehicles = list(world.get_actors().filter("harplab.dreyevr_vehicle.*"))
    if len(ego_vehicles) >= 1:
        DReyeVR_vehicle = ego_vehicles[0]  # TODO: support for multiple ego vehicles?
    else:
        model: str = "harplab.dreyevr_vehicle.model3"
        print(f'No EgoVehicle found, spawning one: "{model}"')
        bp = world.get_blueprint_library().find(model)
        transform = world.get_map().get_spawn_points()[0]
        DReyeVR_vehicle = world.spawn_actor(bp, transform)
    return DReyeVR_vehicle

def find_ego_sensor(world: carla.libcarla.World) -> Optional[carla.libcarla.Sensor]:
    def get_world_sensors() -> list:
        return list(world.get_actors().filter("harplab.dreyevr_sensor.*"))

    ego_sensors: list = get_world_sensors()
    if len(ego_sensors) == 0:
        # no EgoSensors found in world, trying to spawn EgoVehicle (which spawns an EgoSensor)
        if find_ego_vehicle(world) is None:  # tries to spawn an EgoVehicle
            raise Exception(
                "No EgoVehicle (nor EgoSensor) found in the world! EgoSensor needs EgoVehicle as parent"
            )
    # in case we had to spawn the EgoVehicle, this effect is not instant and might take some time
[... 76 lines cut ...]
        denom = d2121 * d4343 - d4321 * d4321
        if abs(denom) < epsilon:
            return 1.0  # no intersection, would cause div by 0 err (potentially)
        numer = d1343 * d4321 - d1321 * d4343

        # calculate scalars (mu) that scale the unit direction XDir to reach the desired points
        # variable scale of direction vector for LEFT ray
        muL = numer / denom
        # variable scale of direction vector for RIGHT ray
        muR = (d1343 + d4321 * (muL)) / d4343

        # calculate the points on the respective rays that create the intersecting line
        ptL = L0 + muL * LDir  # the point on the Left ray
        ptR = R0 + muR * RDir  # the point on the Right ray

        # calculate the vector between the middle of the two endpoints and return its magnitude
        # middle point between two endpoints of shortest-line-segment
        ptM = (ptL + ptR) / 2.0
        oM = (L0 + R0) / 2.0  # midpoint between two (L & R) origins
        FinalRay = ptM - oM  # Combined ray between midpoints of endpoints
        # returns the magnitude of the vector (length)
        return np.linalg.norm(FinalRay) / 100.0

```
i went back to the first fix you told me, and it works with that

GustavoSilvera, 2023-04-26:
Interesting, in all my testing this worked better becauase there was some delay in spawning the EgoVehicle, can you make sure the `find_ego_sensor` is executing correctly and waiting the appropriate time (in this case 10 seconds for the EgoVehicle to spawn) by placing a `print("tick")` in the `while len(ego_sensors) == 0 and time.time() - start_t < maximum_wait_sec` loop?

The logical behavior would be to try to spawn the EgoVehicle (so you should have seen a `"No EgoVehicle found, spawning one"` message being printed, then it should tick for some time to wait for the server to update the actor list and finally grab the corresponding `harplab.dreyevr_sensor.ego_sensor` that should have spawned with the EgoVehicle. 

Do you also have the CarlaUE4.log logfile? This also prints every DReyeVR actor that was spawned and there could have possibly been something wrong indicated there.

closed 2024-01-05 by ajdroid