HARPLab/DReyeVR #113: make package fails after merging with dev branch

Dreyevr, 2023-05-15:
Hi, I saw that you posted a pull request with some changes that looked important, so I decided to pull them by doing this:
```
git pull origin dev
```
After this I executed make install Carla.. ... and then make pythonAPI and make launch, all of which ran succesfully.

However make package give me the following error:
```
 LogInit: Display: Warning/Error Summary (Unique only)
  LogInit: Display: -----------------------------------
  LogInit: Display: LogDReyeVR: Error: [ConfigFile.h::ReadFile:184] Unable to open the config file "C:/CarlaDReyeVR/carla/Unreal/CarlaUE4/Config/EgoVehicles/model3.ini"
  LogInit: Display: LogConsoleManager: Warning: Setting the console variable 'r.Streaming.PoolSize' with 'SetByScalability' was ignored as it is lower priority than the previous 'SetByProjectSetting'. Value remains '4000'
  LogInit: Display: LogModuleManager: Warning: ModuleManager: Unable to load module 'ExampleDeviceProfileSelector'  - 0 instances of that module name found.
  LogInit: Display: LogScript: Warning: Script Msg: A null object was passed as a world context object to UEngine::GetWorldFromContextObject().
  LogInit: Display: LogTemp: Warning: LogitechWheelPlugin initiated!
  LogInit: Display:
  LogInit: Display: Failure - 3 error(s), 9 warning(s)
  LogInit: Display:

  Execution of commandlet took:  159.75 seconds
  LogShaderCompilers: Display: Shaders left to compile 0
  LogTemp: Warning: LogitechWheelPlugin shut down!
  LogHttp: Display: cleaning up 0 outstanding Http requests.
  LogContentStreaming: Display: There are 1 unreleased StreamingManagers
Took 178.2088464s to run UE4Editor-Cmd.exe, ExitCode=1
ERROR: Cook failed.
       (see C:\CarlaDReyeVR\UnrealEngine\Engine\Programs\AutomationTool\Saved\Logs\Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
BUILD FAILED

-[Package]: [ERROR] There was a problem while packaging Unreal project.
          [ERROR] Please read the screen log for more information.
make: *** [package] Error 1
```
The problem lies with the model3.ini file, which is actually called TeslaM3.ini . I tried to change the name of the file, but that interfered with other commands and gave more errors. I am not sure how to edit the command that requires model3.ini file to change it to TeslaM3.ini .

Should I revert back to the previous state? Or could you help me fix this issue if the issue isn't too complicated?

GustavoSilvera, 2023-05-15:
Hi, please don't rename the files. Everything is working as intended and the *old* file is model3.ini while the new file is TeslaM3.ini (since this pr includes the ability to use more than just one ego vehicle).

This error is happening because of a conflict between old files (that haven't yet been deleted) and new files that you just installed, specifically the old model3 blueprint directory has been revamped in the new version. You should first check out the "Upgrading DReyeVR" to run a `make clean` and then reinstall.

Dreyevr, 2023-05-15:
Thanks, that worked.

closed 2023-05-15 by Dreyevr