HARPLab/DReyeVR #114: How to specify 2 arguments simultaneously for make import

Dreyevr, 2023-05-15:
In the carla documentation, there are 2 ways to execute make import
```
make import  ARGS="--package=<package_name>"
```
and
```
make import  ARGS="--no-carla-materials"
```
But how are you supposed to ingest a map with a custom name, and without carla materials at the same time?

GustavoSilvera, 2023-05-15:
I believe just a space between the two arguments will work.

Dreyevr, 2023-05-15:
After importing a map and opening it with the UE4editor, I am prompted with the message that 2 source files have changed and if i want to import them. 

One file is the "own_name_Package", while the other is config\roadpainter_decals and you're asked to choose dataTableRowType (see image below). Whatever i choose, i get the message that the JSON file failed to parse. We usually just ignore this, but I was wondering if this would have some effect at the quality of the textures or the performance in any way. 

Have you ever come across something like this?

GustavoSilvera, 2023-05-15:
No, I have never seen this as I've only ever tested and used the original TownX maps from Carla

closed 2023-06-20 by ajdroid

FBanani, 2023-06-29:
> After importing a map and opening it with the UE4editor, I am prompted with the message that 2 source files have changed and if i want to import them.
>
> One file is the "own_name_Package", while the [...]

Did you solve this issue. I have this problem. Also, in unreal engine nothing is shown?
Could you help me?

elnazbanani, 2023-06-29:
> Did you solve this issue. I have this problem. Also, in unreal engine nothing is shown? Could you help me?

ffff