HARPLab/DReyeVR #115: make: Nothing to be done for 'PythonAPI'

Nitro60zeus, 2023-05-19:
Hi guys!

I have been following the tutorial step by step. In the step instructing to build PythonAPI in conda environment, I am not able to build PythonAPI. I have attached the screenshot as well. My imports (i.e., from DReyeVR_utils and from scenario_runner) passed without any errors/warnings.

[image]

Any help will be greatly appreciated!! Thanks in advance! <3

GustavoSilvera, 2023-05-19:
Seeing as you already have carla, DReyeVR_utils, and scenario-runner working (passed sanity check) in your python install (on `carla13` conda) it seems to me that you've already build PythonAPI and it went successfully? Can you verify that the .egg file is present at `CARLA_0.9.13/PythonAPI/carla/dist/`? Then you should be good to go.

Nitro60zeus, 2023-05-19:
There's only a .whl file. No .egg file. SS atcahed:
[image]

GustavoSilvera, 2023-05-19:
Interesting, usually both are created at a time but this may be OK. Are you having any trouble with your PythonAPI experience? (I would think that since you passed the sanity checks with the `.whl` then you should be fine?). Otherwise you can always try cleaning (`make clean`) and rebuilding (`make PythonAPI`) again.

Nitro60zeus, 2023-05-20:
Ah I see, I see. I tried to run DreyeVR, but I couldn't find or trace the path mentioned in here:

[image]

These are all the folders I have:

[image]

GustavoSilvera, 2023-05-20:
Did you build Carla from source? This looks like just a packaged build. Where is your Build/ directory?

Nitro60zeus, 2023-05-20:
Yep, it's a packaged build. Runs absolutely fine though. That shouldn't affect DReyeVR, right? 

I remember trying to build Carla, but there were some issues which contradicted one another. So I had to go with this packaged build as my last option.

GustavoSilvera, 2023-05-20:
Well, running a packaged build means that you can't install DReyeVR over it (since DReyeVR needs to be built from source. So I'm not sure how you are getting DReyeVR on your system?

Nitro60zeus, 2023-05-20:
Ahh, i see... I shall give it one more try, from the start including building carla. I have a feeling it will work this time. Any advice before I start (particularly any python version you prefer)?

Thanks a lot for your advice. I shall update you soon.

GustavoSilvera, 2023-05-21:
Make sure to closely follow the documentation on building carla, thats about it. There are also plenty of issues and FAQ pages around here that should help you in case you run into trouble.

Nitro60zeus, 2023-05-21:
For sure! Thanks, man.

GustavoSilvera, 2023-05-23:
Can you provide the entire build log, that Xerces build issues should be mostly resolved in the latest version. Have you double checked that you satisfy all the prerequisites from Carla's 0.9.13 windows build instructions?

Nitro60zeus, 2023-05-23:
Yeah yeah, i do satisfy all prerequisites as per the build instructions.

Here's the log

-- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/TrafficManagerLocal.cpp
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/TrafficManagerLocal.h
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/TrafficManagerRemote.cpp
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/TrafficManagerRemote.h
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/TrafficManagerServer.h
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/VehicleLightStage.cpp
  -- Installing: C:/carla/PythonAPI/carla/dependencies/include/carla/trafficmanager/VehicleLightStage.h
  -- Installing: C:/carla/PythonAPI/carla/dependencies/lib/carla_client.lib
  -- Installing: C:/carla/PythonAPI/carla/dependencies/lib/carla_client_debug.lib
-[BuildLibCarla]: LibCarla server has been successfully installed in "C:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "C:\carla\PythonAPI\carla\dependencies\"!
-[BuildOSM2ODR]: [Batch params]: --build --all
Cloning into 'C:\carla\Build\om2odr-source'...
remote: Enumerating objects: 1438380, done.
remote: Counting objects: 100% (13071/13071), done.
remote: Compressing objects: 100% (1772/1772), done.
remote: Total 1438380 (delta 12401), reused 11299 (delta 11299), pack-reused 1425309Receiving objects: 100% (1438380/1438380), 1.01 GiB | 7.02 MiB/s
Receiving objects: 100% (1438380/1438380), 1.01 GiB | 4.53 MiB/s, done.
Resolving deltas: 100% (1226902/1226902), done.
Updating files: 100% (108359/108359), done.
Updating files: 100% (103225/103225), done.
Note: switching to 'ee0c2b9241fef5365a6bc044ac82e6580b8ce936'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at ee0c2b9241f Removed debug warnings
-- Setting build type to 'Release' as none was specified.
CMake Warning (dev) at CMakeLists.txt:23 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.29.30147.0
-- The CXX compiler identification is MSVC 19.29.30147.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR: C:/carla/Build/osm2odr-visualstudio
-- CMAKE_SOURCE_DIR: C:/carla/Build/om2odr-source
--
-- Platform:
--     Host: Windows-10.0.22621 AMD64
--     Target: Windows-10.0.22621 AMD64
--     CMake: 3.26.0-rc2
[... 28 lines cut ...]
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
creating build
creating build\lib.win-amd64-3.7
creating build\lib.win-amd64-3.7\carla
copying source\carla\command.py -> build\lib.win-amd64-3.7\carla
copying source\carla\__init__.py -> build\lib.win-amd64-3.7\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-3.7
creating build\temp.win-amd64-3.7\Release
creating build\temp.win-amd64-3.7\Release\source
creating build\temp.win-amd64-3.7\Release\source\libcarla
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python37\include" "-IC:\Program Files\Python37\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\ [... cut ...] TH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
C:\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: 'OSM2ODR.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist"!

C:\carla>

GustavoSilvera, 2023-05-23:
Can you try cleaning the Xerces build and install folders and providing the logs of `make setup` (just builds all the Carla prerequisites)

Nitro60zeus, 2023-05-23:
OK..
So i have to just type `make setup` in the carla root folder after deleting xerces build & install files. right?

GustavoSilvera, 2023-05-23:
Yep, give that a try

Nitro60zeus, 2023-05-23:
C:\carla>make setup
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  20
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "C:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "C:\carla\Build\" --zlib-install-dir "C:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "C:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...                                                                                       -[Setup]: Installing rpclib...                                                                                              -[install_rpclib]: [Batch params]: --build-dir "C:\carla\Build\"                                                        -[install_rpclib]: A rpclib installation already exists.                                                                -[install_rpclib]: Delete "C:\carla\Build\rpclib-install\" if you want to force a rebuild.                              -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "C:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "C:\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "C:\carla\Build\" --toolset msvc-14.2 --version 1.72.0 -j 20
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "C:\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 file(s) copied.
        1 file(s) copied.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_xercesc]: Retrieving xerces-c.
    -[install_xercesc]: Extracting xerces from "xerces-c-3.2.4-src.zip".
    -[install_xercesc]: Removing "xerces-c-3.2.4-src.zip"
    -[install_xercesc]: Removing dir "C:\carla\Build\manifest"
The system cannot find the file specified.
    -[install_xercesc]: Renaming dir C:\carla\Build\xerces-c-3.2.4 to xerces-c-3.2.4-source
    -[install_xercesc]: Creating "C:\carla\Build\xerces-c-3.2.4-source\build"
    -[install_xercesc]: Creating "C:\carla\Build\xerces-c-3.2.4-install\lib"
    -[install_xercesc]: Creating "C:\carla\Build\xerces-c-3.2.4-install\include"
-- Configuring Apache Xerces-C++ version 3.2.4
-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.22621.
-- The C compiler identification is MSVC 19.29.30147.0
-- The CXX compiler identification is MSVC 19.29.30147.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test CXX_FLAG_W3
-- Performing Test CXX_FLAG_W3 - Success
-- Looking for C++ include arpa/inet.h
-- Looking for C++ include arpa/inet.h - not found
-- Looking for C++ include dlfcn.h
-- Looking for C++ include dlfcn.h - not found
-- Looking for C++ include fcntl.h
-- Looking for C++ include fcntl.h - found
-- Looking for C++ include float.h
-- Looking for C++ include float.h - found
-- Looking for C++ include langinfo.h
-- Looking for C++ include langinfo.h - not found
-- Looking for C++ include limits.h
[... 2498 lines cut ...]
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XML256TableTranscoder.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XML88591Transcoder.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLAbstractDoubleFloat.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLASCIITranscoder.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLBigDecimal.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLBigInteger.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLChar.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLChTranscoder.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLDateTime.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLDOMMsg.hpp
[... cut ...]
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLIBM1047Transcoder.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLIBM1140Transcoder.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLInitializer.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLInteger.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLMutexMgr.hpp
  -- Installing: C:/carla/Build/xerces-c-3.2.4-install/include/xercesc/util/XMLMsgLoader.hpp
  -- Install

GustavoSilvera, 2023-05-23:
Interesting... It looks like it built with no errors? Can you try the same for osm2odr? (This usually also has related issues)

GustavoSilvera, 2023-05-23:
Okay then, can you run make clean and make PythonAPI and save the log again?

Nitro60zeus, 2023-05-23:
Sure. Giving it a shot.

Nitro60zeus, 2023-05-23:
[Make_pythonapi_log.txt](https://github.com/HARPLab/DReyeVR/files/11544384/Make_pythonapi_log.txt)

This is the ,maximum part of the log I am able to save. Might go for a screen recording if this doesn't serve.

Nitro60zeus, 2023-05-23:
Okay, here's the full recording.

https://youtu.be/0CS4US77T0I

GustavoSilvera, 2023-05-23:
I think you should be able to pipe the output into a file by doing something like `make PythonAPI > log.txt`

Nitro60zeus, 2023-05-23:
Here's the file. 
[log.txt](https://github.com/HARPLab/DReyeVR/files/11545813/log.txt)

GustavoSilvera, 2023-05-23:
Okay so it seems to be complaining about a missing project file (visual studios .sln file) for osm2odr, can you check what is inside your osm2odr-visualstudio directory? (Or try deleting the build/source folders and rebuilding and saving the log to a file)

Nitro60zeus, 2023-05-23:
So I deleted the build and source folders for osm2odr and rebuild again. Here's the log file.
[osm2odr_log.txt](https://github.com/HARPLab/DReyeVR/files/11547095/osm2odr_log.txt)

GustavoSilvera, 2023-05-24:
This error message is not very helpful. But just out of curiosity, are you building vanilla carla first or have you installed DReyeVR (make install CARLA=...) over it and then built?

Nitro60zeus, 2023-05-24:
Building Vanilla Carla first

Nitro60zeus, 2023-05-24:
Shall I install DReyeVR and then try to build it?

Nitro60zeus, 2023-05-25:
Update:

So I removed everything and started from scratch.

First Cloned DReyeVR
Then cloned Carla

Installed DReyeVR 1st (make install carla)

Then ran `make pythonAPI`

It went smooth! No issues.

But when I ran `make launch`, the UE4 editor crashed saying 'Fatal error'  :/
And `make package` also ended up with error because of that :/ (log file attached)

Nitro60zeus, 2023-05-25:
Is it because I didn't connect the VR headset? I was doing this without connecting any headsets and steering wheel+pedals.

GustavoSilvera, 2023-05-25:
Are you on the right version of Unreal? It needs to be the specific commit we posted on the documentation (you may need to `git checkout` and rebuild the editor).

Nitro60zeus, 2023-05-25:
Yeah, I am on 4.26. I used the one mentioned in the build instructions: 
[image]

Hmm, lemme try doing git checkout and rebuild editor. So like, do I have to clean carla and dreyevr and start fresh?

Nitro60zeus, 2023-05-25:
Also this, am I doing something wrong?

[image]
[image]

GustavoSilvera, 2023-05-25:
You can try to reboot or manually delete this C:/UnrealEngine/.git/index.lock file to work around this error. You shouldn't need to rebuild Carla/DReyeVR, just the engine.

Nitro60zeus, 2023-05-25:
Ah I see! Thanks. Lemme give it a shot and update you.

Nitro60zeus, 2023-05-25:
update:

So, I followed everything this time. 

Made sure the commit is right, and worked out `git hard reset`
The one needed was at the top 

Had to replace **Commit.gitdeps.xml** for 4.26 from https://github.com/EpicGames/UnrealEngine/tree/release in order to avoid **_download error (403)Forbidden. (Webexception)_**

Then I build it as per instructions and it built successfully.

Then, i did` make launch`, stuff  happened and then right before editor gets launched, it asked me open/load a .render.... file; for which i selected cancel. The editor started to load, but this time it crashed at 75%. 

Then when i tried `make package`, Cook failed. It said something like this  

 **Error: CDO Constructor (LoadAssetMaterialsCommandlet): Failed to find Blueprint'/Game/Carla/Blueprints/LevelDesign/RoadPainterPreset.RoadPainterPreset'**

Full log for `make package`
[Log.txt](https://github.com/HARPLab/DReyeVR/files/11569596/Log.txt)

GustavoSilvera, 2023-05-26:
Okay looks like progress. First, you should verify that the UnrealEngine version is the [CarlaUnreal](https://github.com/carlaunreal/unrealengine) fork version, not the EpicGames version (but since you compiled this should be fine). Next, I believe you might need to re-run the `Update.bat` script in Carla to redownload these assets, check if the filepath at `Unreal/CarlaUE4/Content/Carla/Blueprints/...` exists and contains:
```bash
Game  LevelDesign  Lights  Props  Tools  TrafficLight  Vehicles  Walkers  Weather
```

Nitro60zeus, 2023-05-26:
Update:

IT WORKED!

So I ran `update.bat` and did `make launch` again, and the editor ran perfectly!
Similarly `make package` also went smoothly.

Ran the software on flat screen, looks great! (Except idk why my car is going at 1kmph while on keyboard x'D. Will checkout on steering wheel and pedal asap)

Thanks a ton for guiding and helping me out! Thanks for your time and patience!! Truly appreciate it!

My Summary:
Steps (clone all 3 in one folder/directory):
- Clone DReyeVR
- Clone Carla
- Clone Unreal  (make sure to get the right commit. Check [this](https://github.com/HARPLab/DReyeVR/issues/90#issuecomment-1472260139l)) 
- Build Unreal Engine (if setup.bat fails for unreal engine and you get this:  **_download error (403)Forbidden. (Webexception)_** , then replace `Commit.gitdeps.xml` for version 4.26 from [here](https://github.com/EpicGames/UnrealEngine/tree/release) in order to avoid it. You can select the version from the dropdown menu upon clicking 'release') 
- Install the SRanipal and Logitech Wheel plugins (check [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md))
- Install DReyeVR on Carla first ( in DReyeVR directory, run `make install CARLA=../carla SR=../scenario_runner` )
- Run `Update.bat` (in Carla directory)
- Build Carla (in Carla directory)
- `make PythonAPI`
 - `make launch`
 - `make package`
 - `make check`

Nitro60zeus, 2023-05-31:
btw @GustavoSilvera i was trying to run DReyeVR_logging.py, i got this error:
[image]

I started the Ue4carla, then added traffic and then tried to run DReyeVR_logging.py
Am i missing a step in between?

GustavoSilvera, 2023-05-31:
When you start the server do you have an EgoVehicle already present? It looks like it is trying to spawn one because it can't find one, but this spawning seems to fail. 

Also can you try the `dev` branch which may have a fix for this. (Try it in a separate installation to not affect your current one.

GustavoSilvera, 2023-05-31:
Yes, you can use the same `UE4_ROOT`. The `dev` branch only affects the Carla installation.

Nitro60zeus, 2023-06-02:
OK, so i did this. Looks like it worked... Like a huge amount of stuff appeared rapidly which i guess to be the eye data. Does it get stored automatically or do we have to store manually? Something like this: python DReyeVR_logging.py > log.txt

Nitro60zeus, 2023-06-02:
Update: Tried to log the data. Did:
 >carlaue4.exe -vr
 >ran a scenario
 >python DReyeVR_logging.py > a1.txt 

Got this:
[image]

Need a little help comprehending it ^^". Plus, i think something is wrong in the data... Like every eye data is literally 0. I am missing something for sure ig..

GustavoSilvera, 2023-06-02:
Yes, this is the eye data that is logged from the server. It can be saved to a file as you described above, or you can run a recording and it will automatically save all the data into a format that can be easily parsed by our [DReyeVR parser](https://github.com/harplab/dreyevr-parser) (with the added benefit that you can then replay/re-enact the events. 

Since the eye data is 0 this means there is no correct eye information. You can tell because all the `*_valid` variables are booleans indicating whether the corresponding data is valid or not, and since they are all `False` this indicates that the eye tracker is not working correctly (so all the values are default). Can you see if the eye tracker is working? (is the reticle in the server moving around?)

Nitro60zeus, 2023-06-02:
Ah, I see.

Yeah, the eye tracker is working perfectly. The reticle points at things the observer looks at inside the VR with 100% accuracy.

GustavoSilvera, 2023-06-02:
Interesting, can you try a recording?

GustavoSilvera, 2023-06-06:
How are you running a scenario? If you are using our `run_experiment.py` script (in our ScenarioRunner fork) then it will automatically begin recording for you. Then, once the scenario is completed (not while it is still running), restart the DReyeVR server and run the `start_replaying.py` with the corresponding file. 

The `utils` import issue does seem like a bug. Can you manually add this path to your PYTHONPATH as a workaround for now?

Nitro60zeus, 2023-06-07:
Yeah yeah, i was using `run_experiment.py` along with an example route. 

Ok, so i did the following:
1. opened dreyevr
2.  ran `python run_experiment.py --title "dreyevr_experiment" --route SCENARIO_RUNNER_ROOT%\srunner\data\routes_debug.xml %SCENARIO_RUNNER_ROOT%\srunner\data\all_towns_traffic_scenarios1_3_4.json 0`
3. closed both
4. started dreyevr again
5. in a new x64 cmd prompt:
[image]

Nitro60zeus, 2023-06-07:
Also added the PYTHONPATH in environment variable:
[image]

Something like this shows up:
[image]

GustavoSilvera, 2023-06-07:
Ah, well in this case you should just install `pandas` using either `pip` or `conda` (if you are in a conda environment)

Nitro60zeus, 2023-06-08:
Cools,
So i installed pandas, here's the update:

>C:\DReyeVR-parser>py example.py --file C:\carla\PythonAPI\examples\recorder4.txt
Did not find data at cache\recorder4.pkl
Reading DReyeVR recording file: C:\carla\PythonAPI\examples\recorder4.txt
successfully read 1170 frames in 4.371s
cached data to cache\recorder4.pkl
output figure to results/left_pupil_diameter_vs_time.png
output figure to results/right_pupil_diameter_vs_time.png
Total validity percentage: 29.658%
output figure to results/confidence_(validity)_vs_time.png
ERROR: some gaps still unfilled!!
output figure to results/left_pupil_diameter_vs_time.png
ERROR: some gaps still unfilled!!
output figure to results/right_pupil_diameter_vs_time.png
output figure to results/yaw_c_x_pitch_c_histogram2d.png
output figure to results/yaw_l_x_pitch_l_histogram2d.png
output figure to results/yaw_r_x_pitch_r_histogram2d.png
output figure to results/egopos_xyz.png
output figure to results/egopos_xyz.png
output figure to results/egorot_pyr.png
output figure to results/vehicle_position_over_time.png
output figure to results/left_pupil_position.png
output figure to results/lpupilx_x_lpupily_histogram2d.png
output figure to results/right_pupil_position.png
output figure to results/rpupilx_x_rpupily_histogram2d.png
output figure to results/left_eye_openness_vs_time.png
output figure to results/right_eye_openness_vs_time.png
output figure to results/ego_speed_vs_time.png
output figure to results/ego_accel_vs_time.png
output figure to results/smooth_ego_accel_vs_time.png
output figure to results/delta_egorot_pyr.png
output figure to results/steering_vs_time.png
output figure to results/egovel_xyz.png
output figure to results/egoaccel_xyz.png
output figure to results/egorot_xyz.png
output figure to results/velocity_vs_time.png
output figure to results/throttle_vs_time.png
output figure to results/brake_vs_time.png
output figure to results/cameraloc.png
output figure to results/camerarot.png
output figure to results/combinedgaze.png
output figure to results/leftgaze.png
output figure to results/rightgaze.png
output figure to results/actor_id_302_position_over_time.png
Minimum distance: 3.90m
output figure to results/distance_to_actor_302_vs_time.png
output figure to results/actor_id_148_position_over_time.png
Minimum distance: 113.02m
output figure to results/distance_to_actor_148_vs_time.png
Traceback (most recent call last):
  File "example.py", line 524, in <module>
    main(args.file, args.out)
  File "example.py", line 471, in main
    pos3D = actor_data["Location"][non_zeros]
IndexError: arrays used as indices must be of integer (or boolean) type
>

GustavoSilvera, 2023-06-08:
Yep, you seem to have parsed the data alright. Take a look at the results. There seems to be some errors at the end though. Not sure whats up with that. Also if you are now changing the discussion to the parser, please create a new thread as this one is far too bloated.

Nitro60zeus, 2023-06-09:
Yeah, i see some graphs. I thought parsing failed. 

Oh, sorry for using the same thread. I shall close this and use a new one.

Thanks a ton for all the help <3

closed 2023-06-09 by Nitro60zeus

Nitro60zeus, 2023-06-09:
> How are you running a scenario? If you are using our `run_experiment.py` script (in our ScenarioRunner fork) then it will automatically begin recording for you. Then, once the scenario is completed [...]

Just one last question. Here, when you say to use start_replaying.py with the corresponding file (i guess you are talking about the .rec file), could u plz show an example? Assume the name of the file be 'test1.rec'. So what will be the exact command in cmd?

GustavoSilvera, 2023-06-10:
Yes the .rec is a binary recording file used by Carla. So the appropriate syntax would be:

`start_replaying.py -f test.rec`