HARPLab/DReyeVR #116: FileNotFound error when running the scenario runner

zzhzz4429, 2023-05-30:
Hello,
I was tried to run the run_experiment.py with this command, and it gave me the error output below, (I used this fork for scenario runner [https://github.com/harplab/scenario_runner/commits/DReyeVR-0.9.13](url),) 
[image]
However I checked my data folder and found that the all_route_signs.json did exist
[image]
Does anyone have any good suggestion on this problem? Thank you for your patience!

GustavoSilvera, 2023-05-30:
Not sure, it is probably something with the relative paths being messed up for some reason. You can try to fix this by using an absolute path for the file. 

But also, instead of using this file (which hardcodes all the sign positions, for routes that have been hand-curated) we'd recommend you instead to use the [automatic sign placement](https://github.com/HARPLab/scenario_runner/commit/183920d2f34ecb03d0a67a46a2d76635a40c070b) that was introduced into the scenario-runner DReyeVR branch recently. If you are on the latest (`DReyeVR-0.9.13`) branch of our ScenarioRunner fork then it should be used automatically and you can completely ignore the direction-sign file (just comment it out if its giving you trouble).

GustavoSilvera, 2023-05-31:
The error happens because `The scenario cannot be loaded` for some reason. Are you sure this scenario configuration you have is valid? (`route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 9`)

GustavoSilvera, 2023-05-31:
If you try this command, does it work or fail similarly?
```
python scenario_runner.py  --route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 10 --output --timeout 1000 --reloadWorld --sync
```

I was able to pull this scenario from [an existing comment thread](https://github.com/HARPLab/DReyeVR/issues/96#issuecomment-1489002872), which seemed to work for others.

zzhzz4429, 2023-06-01:
It still failed and throw the previous errors.
```
(carla12) C:\VR_testbed\scenario_runner>python scenario_runner.py  --route srunner/data/routes_training.xml srunner/data/all_towns_traffic_scenarios.json 10 --output --timeout 1000 --reloadWorld --sync
scenario_runner.py:98: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.12'):
INFO:  Found the required file in cache!  Carla/Maps/TM/Town03.bin
Preparing scenario: RouteScenario_10
Spawning (static.prop.dreyevr_sign_straight) sign 21 at Transform(Location(x=-20.366112, y=210.581940, z=2.000000), Rotation(pitch=360.000000, yaw=89.857056, roll=0.000000))
Spawning (static.prop.dreyevr_sign_left) sign 365 at Transform(Location(x=246.653214, y=80.986641, z=2.082980), Rotation(pitch=359.527618, yaw=1.393250, roll=0.000000))
Spawning (static.prop.dreyevr_sign_straight) sign 432 at Transform(Location(x=185.645752, y=54.999397, z=2.000000), Rotation(pitch=0.000000, yaw=269.852600, roll=0.000000))
Spawning (static.prop.dreyevr_sign_straight) sign 516 at Transform(Location(x=105.771606, y=53.487377, z=2.000000), Rotation(pitch=0.000000, yaw=-90.147385, roll=0.000000))
Spawning (static.prop.dreyevr_sign_right) sign 616 at Transform(Location(x=94.828636, y=70.091522, z=2.000000), Rotation(pitch=0.000000, yaw=53.550362, roll=0.000000))
Spawning (static.prop.dreyevr_sign_right) sign 678 at Transform(Location(x=150.974686, y=66.288628, z=2.000000), Rotation(pitch=360.000000, yaw=89.852615, roll=0.000000))
Spawning (static.prop.dreyevr_sign_straight) sign 895 at Transform(Location(x=21.138271, y=126.940002, z=2.000000), Rotation(pitch=360.000000, yaw=269.182800, roll=0.000000))
Spawning (static.prop.dreyevr_sign_right) sign 978 at Transform(Location(x=-61.208221, y=126.352516, z=2.000000), Rotation(pitch=360.000000, yaw=268.703186, roll=0.000000))
Spawning (static.prop.dreyevr_sign_left) sign 1091 at Transform(Location(x=-73.937134, y=28.531349, z=2.087857), Rotation(pitch=359.246429, yaw=-0.156219, roll=0.000000))
Spawning (static.prop.dreyevr_sign_left) sign 1169 at Transform(Location(x=-134.689957, y=-7.159969, z=2.000000), Rotation(pitch=0.000000, yaw=-449.705322, roll=0.000000))
Spawning (static.prop.dreyevr_sign_goal) sign 1203 at Transform(Location(x=-154.519943, y=22.429756, z=2.000000), Rotation(pitch=0.000000, yaw=180.029846, roll=0.000000))
The scenario cannot be loaded
Traceback (most recent call last):
  File "scenario_runner.py", line 393, in _load_and_run_scenario
    debug_mode=self._args.debug)
  File "C:\VR_testbed\scenario_runner\srunner\scenarios\route_scenario.py", line 171, in __init__
    world, ego_vehicle, self.config, self.route, timeout=self.timeout))
NameError: name 'ego_vehicle' is not defined
name 'ego_vehicle' is not defined
Exception ignored in: <function RouteScenario.__del__ at 0x000001EF1C247730>
Traceback (most recent call last):
  File "C:\VR_testbed\scenario_runner\srunner\scenarios\route_scenario.py", line 605, in __del__
    self.remove_all_actors()
  File "C:\VR_testbed\scenario_runner\srunner\scenarios\basic_scenario.py", line 203, in remove_all_actors
    for i, _ in enumerate(self.other_actors):
AttributeError: 'RouteScenario' object has no attribute 'other_actors'
No more scenarios .... Exiting
```

GustavoSilvera, 2023-06-01:
Interesting. This shouldn't happen. Are you on the latest version of ScenarioRunner (our Fork: DReyeVR-0.9.13) and DReyeVR? Have you tried building a new instance (of Carla + DReyeVR, you can keep the same UE4) with the `dev` changes (which we will merge into main soon)

zzhzz4429, 2023-06-01:
Sounds worth a shot. But do I need to make pythonAPI before the DReyeVR installation and then make PythonAPI again? Because usually there are lots of issues of the origin windows build pipeline.
[image]

GustavoSilvera, 2023-06-01:
You can copy over the Build/ folder from one install to the other, so most of the windows build issues should not happen again. But yes, you should rebuild the PythonAPI since we made (slight) changes to it

zzhzz4429, 2023-06-01:
Thanks for the advice. I do have another question now, I cloned the latest version of scenario runner from [your fork](https://github.com/HARPLab/scenario_runner.git). After I run make install in the DReyeVR folder it showed message like this
[image]
should I proceed anyways? Just tried to make sure every steps is correct.

GustavoSilvera, 2023-06-01:
you should switch the scenario runner to have the `v0.9.13` git branch

zzhzz4429, 2023-06-01:
It worked! Thank you for your time and patience!
[image]

closed 2023-06-01 by zzhzz4429