HARPLab/DReyeVR #117: Cannot find vehicle config (.ini) files when adding my own EgoVehicle

LowP1tch, 2023-06-05:
Hi Guys!

I am currently triying to exchange the default model3 with a 2wheeled vehicle from the carla blueprints. I am at Step 3.2 "Add the new config file and code" of the CustomEgo.md and I am trying to create the new .ini file. Unfortunately the "EgoVehicles" folder is not existend inside of CarlaUE4/Config/. Hence there are also no existend .ini files I could copy and adjust, rename etc. 
Did I do anything wrong when building and launching? Since everything else works just fine it might be an outdated tutorial? Or Is there maybe a chance to download and paste the needed config files?

Some other points that might be interesting:
- In the "DReyeVRFactory.h" was said to add the name of the new egovehicle to the list (Step 3.1). But I had to add the whole list myself before I could add anything. I also found that in the online version (https://github.com/HARPLab/DReyeVR/blob/main/DReyeVR/DReyeVRFactory.h) it says `TMap<FString, UClass *> BP_Classes;` in the last code line where as in my file it says `class UClass *EgoVehicleBPClass = nullptr;`. 
- Also strangely in the documentation it always says "TeslaM3" where as in carla\Unreal\CarlaUE4\Content\DReyeVR\EgoVehicle the blueprint's name is "model3".
- I just finished building and launching last week on Windows 10. Initially I had some probems launching, but "Update.bat" and working in anaconda environment made it work.

I would be so greatfull for any help!

GustavoSilvera, 2023-06-06:
Have you updated to the latest `main` branch? We recently merged a pull request that includes this backend overhaul (#111). 

Please see [this](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Install.md#upgrading-dreyevr) section on how to upgrade.

LowP1tch, 2023-06-19:
Yes this actually solved my problem! Thank you so much. I didn't expect not being on the main branch since my install was still so fresh.

closed 2023-09-21 by ajdroid