HARPLab/DReyeVR #118: DReyeVR Parser data output

Nitro60zeus, 2023-06-09:
Hi guys!

Um, so I ran the parser successfully. But there seem to be some issues.... The eye tracker works absolutely fine with DReyeVR. The output for gaze metrics and eye openess is missing. Here's the pic:
[image]

But it seems to collect pupil diameter:
[image]

Also, i assume the data is being collected correctly, because here it says 1 for `valid`:
[image]

GustavoSilvera, 2023-06-10:
Looks like the data is there but the parser didn't get the data for the gaze data. Can you make sure this is parsed correctly? Try printing the element of the dictionary.

Nitro60zeus, 2023-06-10:
So I tried printing the element "LeftEyeOpenness" and "RightGaze". All values come out as 0. 

Do you mind giving a look at the human readable file?  Would be extremely helpful.  ^^"
Link (expires in 3 days): https://litter.catbox.moe/agbuji.txt

Nitro60zeus, 2023-06-12:
[image]

Been trying to trouble shoot this, but i am not able to understand where exactly is the issue. Tried deleting and re-cloning the parser and started all over again, but still the same issue :/

Files (rec and txt): [test23ax.zip](https://github.com/HARPLab/DReyeVR/files/11726071/test23ax.zip)

The `DReyeVR_logging.py` data looks absolutely great. Is there's any way we can use this or simplify to a more readable representation?
File: [test23.txt](https://github.com/HARPLab/DReyeVR/files/11726095/test23.txt)

GustavoSilvera, 2023-06-13:
Unfortunately this looks like a bug in the (DReyeVR) recorder rather than in the parser. Which is much more difficult to debug. I am away from a computer for a while and will be unable to debug for the time being. Hopefully you can use the raw logging variables for now.

Nitro60zeus, 2023-06-13:
Oh, I see. No problem though, I am working out a way to use the raw logging variables to convert them into csv format. Should serve the purpose ig.

But yeah, thanks a lot man for helping all the way out, this far. Truly appreciate it <3

closed 2023-09-06 by ajdroid

FBanani, 2023-09-21:
> Oh, I see. No problem though, I am working out a way to use the raw logging variables to convert them into csv format. Should serve the purpose ig.
> 
> But yeah, thanks a lot man for helping all the way out, this far. Truly appreciate it <3

Hi I have kind of this issue. However I am in VR mode itshows the gaze and pupil zero!!! COULD YOU HELP ME
[image]

Nitro60zeus, 2023-09-22:
There are two ways you can visualize data. 

1) Using DReyeVR parser: Make sure its calibrated well for the subject using it. If its not calibrated properly, such issues can occur. If it doesn't work, try deleting the parser and loading again. 

2) Using DReyeVR_logging.py: This is great if you wish to manually play around with the data. The output I presume is in python dictionary format. So you  need to find a way to convert into csv for manipulation.

Lemme know how it goes.

FBanani, 2023-09-22:
> There are two ways you can visualize data.
>
> 1. Using DReyeVR parser: Make sure its calibrated well for the subject using it. If its not calibrated properly, such issues can occur. If it doesn't [...]

Thanks for replying.
You mean the VR headset should be calibrate or something else.
Also, when I execute  DReyeVR_logging.py in anaconda prompt I just see some data and I do not know where the data was saved?

Nitro60zeus, 2023-09-22:
Yeah, the headset eye calibration must be done. Also, make sure you are using x64 Native Tools CMD for the scripts. To see the data, you need to jot it down in a .txt file. Below, I have explained how:

DReyeVR_logging.py can be used in two ways:

1) When you are actively running simulation, you simultaneously start the script in another CMD window. For this you just run `py DReyeVR_logging.py > log.txt` in a cmd window.

2) You simultaneously start this script when you are playing a recording (.rec file). For this, you run the logging first, and the recording immediately after that in two separate cmd windows. 

first this `py DReyeVR_logging.py > log.txt` and then this `py show_recorded_file_info.py -a -f /PATH_TO_.rec file` 

The log.txt will appear in the same folder where the DReyeVR.py script is present.