HARPLab/DReyeVR #119: "Fatal Error" while using start_replaying.py

Nitro60zeus, 2023-06-10:
So i followed the format to play the .rec file, but it seems to crash. Here's the pic

[image]

anastaruno, 2023-06-11:
Hi! It is hard to determine the cause of the issue without any additional information. Could you please send the recording file so we can learn more? Meanwhile, try running 'make check' from your DReyeVR directory to ensure it is installed correctly.

Nitro60zeus, 2023-06-12:
Hi! Herein i have attached the .rec file in zip format. Lemme know if there's any issues.

[exp_a1_test_06_09_2023_12_39_33.zip](https://github.com/HARPLab/DReyeVR/files/11724516/exp_a1_test_06_09_2023_12_39_33.zip)

For sure, I shall run `make check` and update you asap.

Nitro60zeus, 2023-06-12:
Update: `make check` was successful. No issues.

shh1v, 2023-07-13:
After getting the fatal error, could you upload the file $CARLA_ROOT$/Unreal/CarlaUE4/Saved/Logs/CarlaUE4.log here?

Nitro60zeus, 2023-07-14:
[CarlaUE4.log](https://github.com/HARPLab/DReyeVR/files/12051364/CarlaUE4.log)

Hey, thanks for your response! Here's the file.

Nitro60zeus, 2023-09-26:
Removed and installed again, and this time I made sure to run script using x64 Native Tools CMD prompt. Worked.

closed 2023-09-26 by Nitro60zeus