HARPLab/DReyeVR #124: I have got a new error while I want to make launch after making PythonAPI

FBanani, 2023-07-26:
[image]
Hi @GustavoSilvera 
I want to install the DReyeVR and when I make launch afet making pythonAPI, I have got this error. Could you help me?
Thanks

ajdroid, 2023-09-06:
What version of the dreyevr code are you using?

FBanani, 2023-09-07:
I do not know.
Also I have another issue:
[image]
I run "python DReyeVR_logging.py," but when my experiment is finished, what should I do to collect the data? I mean, where is it saved?

ajdroid, 2023-09-11:
Can you provide the full paths of your built package and the DReyeVRConfig file you are editing?

FBanani, 2023-09-11:
thanks alot.
do you mean this:
[image]

FBanani, 2023-09-11:
> thanks alot. do you mean this: [image]

FBanani, 2023-09-11:
> Can you provide the full paths of your built package and the DReyeVRConfig file you are editing?

Thanks a lot.
Do you mean this:
[image]

ajdroid, 2023-09-11:
If you want it to change in the package, you should modify the corresponding DReyeVRConfig. What is the path to the binary you are running that is built via make package?

FBanani, 2023-09-11:
> This looks like the config file in the dreyevr source. If you want it to change in the package, you should modify the corresponding DReyeVRConfig. What is the path to the binary you are running that is built via make package?

I modify the code here and change the "DrawFlatReticle=False; reticle in flat-screen mode" to False. then in carla root I make a package.
[image]

FBanani, 2023-09-11:
> This looks like the config file in the dreyevr source. If you want it to change in the package, you should modify the corresponding DReyeVRConfig. What is the path to the binary you are running that is built via make package?

I am really sorry. I am not good at computer since I am civil engineer. In the DReyeVRConfig.h file I just change this line then in X64 in the carla root I execute make package.

ajdroid, 2023-09-11:
In your `carla_root/Build\UE4Carla\0.9.13-*-*\WindowsNoEditor\CarlaUE4\Config` you will see the DReyeVRConfig file that you need to modify in order to make changes to your built version of DReyeVR. You do not need to re execute make package.

FBanani, 2023-09-11:
> In your `carla_root\Build\UE4Carla\0.9.13-*-*\WindowsNoEditor\CarlaUE4\Config` you will see the DReyeVRConfig file that you need to modify in order to make changes to your built version of DReyeVR. You do not need to re execute `make package`.

Thank you so much.
Could you help me how to save the all simulator data as well.

ajdroid, 2023-09-11:
Please read our docs: https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#recordingreplaying-a-scenario

FBanani, 2023-09-11:
> Please read our docs: https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#recordingreplaying-a-scenario

Thanks but I saw in other issue "[How to acquire eye tracking data maybe in a better way?](https://github.com/HARPLab/DReyeVR/issues/81#top)
#81" GustavoSilvera mentioned used this code "python PythonAPI/examples/DReyeVR_logging.py" for gathering better eye gaze. Therefore, I overwhelmed.
Sorry for my basic question.

FBanani, 2023-09-11:
> That is for getting eye tracking data in real time. If you just want to store the experiment to disk and replay or analyze it later, please use the resources in the Usage.md file.

thanks I found it in this directory:
[image]

is it correct

ajdroid, 2023-09-11:
I just meant the resources documented in this link: https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#recordingreplaying-a-scenario

FBanani, 2023-09-11:
> I just meant the resources documented in this link: https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#recordingreplaying-a-scenario

Thanks a lot.

FBanani, 2023-09-21:
Hi @ajdroid  I come back with another question 
I execute python start_recording.py -n 0 in screen mode
[image]
then execute this
python show_recorder_file_info.py -a -f C:\test1.log > recorder.txt
then with DReyeVR-parse
[image]
show me this error. 
Also I have another issue when I run the following after python start_recording.py -n 0 during my experiment it gives me the self error.
[image]

FBanani, 2023-09-21:
Hi @ajdroid I have another issue. However I am in VR mode it shows the gaze and pupil zero!!! COULD YOU HELP ME
[image]

ajdroid, 2023-09-21:
Hi @FBanani please create a new issue if you have an unrelated problem to the main thread title. I am closing this issue.

closed 2023-09-21 by ajdroid