HARPLab/DReyeVR #127: Cannot connect with Steam VR

ALEX95GOGO, 2023-07-30:
Hi, I installed everything sucessfully. But when I launched Carla with ./CarlaUE4.exe -vr, and I opened steam VR but nothing is play there. I can also see the SRanipalRuntime works since the icon turns green. Do you have any suggestions? Thanks a lot!
[image]

ajdroid, 2023-07-31:
Can you use the UE4 editor to do a VR preview? Does that work?

ALEX95GOGO, 2023-08-03:
Hi, I compiled the project in one pc and distribute it to another pc. For the pc I compiled I can do a VR preview, but for the pc I distributed it can open steam VR sometimes, but not always. Do you have any suggestions for this? Thanks a lot for your help!

ALEX95GOGO, 2023-08-29:
I made a new account in Windows and this problem is solved. Maybe it's due to something wrong with my steam VR installation.

closed 2023-09-06 by ajdroid