HARPLab/DReyeVR #128: No HUD in VR mode

ALEX95GOGO, 2023-08-12:
Hi, the HUD in the server looks fine, but not working in VR mode. Is there any configuration I need to change?

GustavoSilvera, 2023-08-15:
I'm not sure what HUD you are referring to but the VR (HMD) display does not have a dedicated HUD. Only the 3D dash elements on the vehicle (ex. Speedometer/gear/etc.). The flat-screen render has a "HUD" that shows fps and other UI elements (see FlatHud.h|cpp) but this is specifically NOT for VR.

ALEX95GOGO, 2023-08-29:
Thanks a lot for your answer. I mean the Speedometer text, and I am also trying to add some customized text by modifying the Egovehicle.cpp. The text I am adding is to display a warning message "Too close" when the ego vehicle is too close to the leading vehicle. I have successfully implemented this by "make launch", however, when I distributed the package by "make package", the .exe file failed to display the warning message. I will appreciate it if you have any suggestions.

ajdroid, 2023-09-06:
Did you use debug variables to implement the "Too close" message? That could be why it wouldn't show up in the packaged simulator

closed 2023-11-11 by ALEX95GOGO