HARPLab/DReyeVR #133: Can I use DReyeVR_logging.py with the recorded file (.rec) ?

Nitro60zeus, 2023-09-13:
Hi!

Is there any way to use the `DReyeVR_logging.py` with the `.rec` file ? Currently, I have to run it with my simulation to log manually.

Thanks.

Nitro60zeus, 2023-09-15:
ok, I figured out a way. Basically, if we run `DReyeVR_logging.py` with `start_replaying.py`, we can get the log file. 

But my question now is, what's the output format of that log file?

ajdroid, 2023-09-21:
Generally, we don't use the `DReyeVR_logging.py` file to get recorded information, rather we recommend using the DReyeVR parser as described [here](https://github.com/HARPLab/DReyeVR/blob/main/Docs/Usage.md#recordingreplaying-a-scenario). What information are you trying to get?

Nitro60zeus, 2023-09-22:
I want to perform some data analysis. For example, I want to see how the average pupil diameter varies for n participants. So I can have n columns containing pupil data for each participant and a last column that contains the average. So yeah, stuff like that. 

Also yeah, the parser works great! Its perfect to analyse data individually. The graphs it gives have the "carla_time" on the x-axis. I am interested in data vs "real world time". Can you lemme know which variable is associated with "real world time" ?

Nitro60zeus, 2023-09-26:
Ok, figured it out. So `'timestamp_device'` is the one associated with real-time. Need to divide by 1000 to get the value in seconds.

closed 2023-09-26 by Nitro60zeus

Nitro60zeus, 2023-10-01:
Update:
An easier way is to run the text file that you get from `./show_recorder_file_info.py -a -f /PATH/TO/RECORDER-FILE > recorder.txt` on this code to get your desired variables. You can modify as per your needs.

```
import re
import csv

# Open the input text file
with open('4a.txt', 'r') as file:
    content = file.read()

# Use regular expressions to find all relevant values
timestamp_carla_values = re.findall(r'TimestampCarla:(\d+)', content)
pupil_diameter_values = re.findall(r'PupilDiameter:(-?[\d.]+)', content)
eye_openness_values = re.findall(r'EyeOpenness:(-?[\d.]+)', content)
hit_values = re.findall(r'FocusInfo:Hit:(\d+)', content)
actor_name_values = re.findall(r'ActorName:([\w]+)', content)

# Open a CSV file for writing
with open('z.csv', 'w', newline='') as csvfile:
    writer = csv.writer(csvfile)

    # Write the header row to the CSV file
    writer.writerow(['TimestampCarla', 'PupilDiameter A', 'PupilDiameter B', 'EyeOpenness A', 'EyeOpenness B', 'Hit', 'ActorName'])

    # Determine the minimum length of the lists to avoid index errors
    min_length = min(len(timestamp_carla_values), len(pupil_diameter_values), len(eye_openness_values), len(hit_values), len(actor_name_values))

    # Write the values to the CSV file, alternating EyeOpenness values
    for i in range(min_length):
        row = [
            timestamp_carla_values[i],
            pupil_diameter_values[i * 2] if i * 2 < len(pupil_diameter_values) else '',
            pupil_diameter_values[i * 2 + 1] if i * 2 + 1 < len(pupil_diameter_values) else '',
            eye_openness_values[i * 2] if i * 2 < len(eye_openness_values) else '',
            eye_openness_values[i * 2 + 1] if i * 2 + 1 < len(eye_openness_values) else '',
            hit_values[i] if i < len(hit_values) else '',
            actor_name_values[i] if i < len(actor_name_values) else ''
        ]
        writer.writerow(row)

print("CSV file 'z.csv' has been created with the specified values.")
```