HARPLab/DReyeVR #134: Carla crashed in VR mode

ALEX95GOGO, 2023-09-14:
The CarlaUE4.exe crashed after a few minutes in VR mode, it works fine without VR but in VR it almost crashed every time. Here is the crash analysis I got from the dmp file:

FILE_IN_CAB: UE4Minidump.dmp

CONTEXT: (.ecxr)
rax=000001d462ecdaa8 rbx=000000b78b7fb3f0 rcx=000001d3884f13cc
rdx=0000000000000004 rsi=0000000000000400 rdi=0000000000000000
rip=00007ffb4213cf19 rsp=000000b78b7fb210 rbp=00007ff6372af648
r8=0000000000000040 r9=0000000100027f64 r10=0000000100000001
r11=0000000000000001 r12=000001d3b581deb0 r13=0000000000000000
r14=000000b78b7fb3f0 r15=00007ff6371d90c0
iopl=0 nv up ei pl nz na pe nc
cs=0033 ss=002b ds=002b es=002b fs=0053 gs=002b efl=00000202
KERNELBASE!RaiseException+0x69:
00007ffb`4213cf19 0f1f440000 nop dword ptr [rax+rax]
Resetting default scope

EXCEPTION_RECORD: (.exr -1)
ExceptionAddress: 00007ffb4213cf19 (KERNELBASE!RaiseException+0x0000000000000069)
ExceptionCode: 00004000
ExceptionFlags: 00000000
NumberParameters: 1
Parameter[0]: 000000b78b7fb310

PROCESS_NAME: CarlaUE4-Win64-Shipping.exe

ERROR_CODE: (NTSTATUS) 0x4000 -

EXCEPTION_CODE_STR: 4000

EXCEPTION_PARAMETER1: 000000b78b7fb310

STACK_TEXT:
000000b78b7fb210 00007ff63319d426 : 000000b78b7fb910 00007ffb3dcfcd43 000000b78b7fb8d0 00007ffb3deba360 : KERNELBASE!RaiseException+0x69
000000b78b7fb2f0 000000b78b7fb910 : 00007ffb3dcfcd43 000000b78b7fb8d0 00007ffb3deba360 000000b78b7fb3f0 : CarlaUE4_Win64_Shipping+0x116d426
000000b78b7fb2f8 00007ffb3dcfcd43 : 000000b78b7fb8d0 00007ffb3deba360 000000b78b7fb3f0 000000b700000002 : 0x000000b78b7fb910 000000b78b7fb300 00007ffb3dd0df54     : 0000000000000000 0000000000000038 000001d3a67102f4 00007ff63303e393 : d3d11!CDevice::CreateLayeredChild+0x263 000000b78b7fb740 00007ff633063a93     : 00007ff635c9f030 00007ff635b94e88 0000000000000000 0000000000000001 : d3d11!ATL::AtlInternalQueryInterface+0xf4 000000b78b7fb780 00007ff635c9f030     : 00007ff635b94e88 0000000000000000 0000000000000001 000000b78b7fb890 : CarlaUE4_Win64_Shipping+0x1033a93 000000b78b7fb788 00007ff635b94e88     : 0000000000000000 0000000000000001 000000b78b7fb890 00007ff63280ffe2 : CarlaUE4_Win64_Shipping+0x3c6f030 000000b78b7fb790 0000000000000000     : 0000000000000001 000000b78b7fb890 00007ff63280ffe2 00000000`00000018 : CarlaUE4_Win64_Shipping+0x3b64e88

SYMBOL_NAME: carlaue4_win64_shipping+116d426

MODULE_NAME: CarlaUE4_Win64_Shipping

IMAGE_NAME: CarlaUE4-Win64-Shipping.exe

STACK_COMMAND: ~89s; .ecxr ; kb

FAILURE_BUCKET_ID: APPLICATION_FAULT_4000_CarlaUE4-Win64-Shipping.exe!Unknown

OS_VERSION: 10.0.19041.1

BUILDLAB_STR: vb_release

OSPLATFORM_TYPE: x64

OSNAME: Windows 10

IMAGE_VERSION: 4.26.2.0

FAILURE_ID_HASH: {4cc2ad21-3bd7-538c-1f10-d7f543dd82e3}

ALEX95GOGO, 2023-09-15:
I found it happens when the participant turns their head, when they are not moving a lot it was fine. Also, turning off all the mirrors can temporally solve this problem. Does it means the game can crash when the fps is too low?

ajdroid, 2023-09-21:
Hi, we have not encountered this problem but we have not tested it on GPUs other than 3090/4090. What is your hardware configuration?

ALEX95GOGO, 2023-09-25:
My GPU is NVIDIA Quadro P5000. Does the crash have something to do with it?

ajdroid, 2023-09-25:
Yes, that GPU cannot handle the graphics load that the simulator will put on it. I would recommend a higher end GPU like the Nvidia RTX 3090, 4080, or 4090

closed 2024-01-05 by ajdroid