HARPLab/DReyeVR #135: getting Data From Simulator

FBanani, 2023-09-21:
Hi @ajdroid  I come back with another question 
I execute python start_recording.py -n 0 in screen mode
[image]
then execute this
python show_recorder_file_info.py -a -f C:\test1.log > recorder.txt
then with DReyeVR-parse
[image]
show me this error. 
Also I have another issue when I run the following after python start_recording.py -n 0 during my experiment it gives me the self error.
[image]

FBanani, 2023-09-21:
Hi @ajdroid I have another issue. However I am in VR mode it shows the gaze and pupil zero!!! COULD YOU HELP ME
[image]

FBanani, 2023-09-21:
Hi @ajdroid 
I have another issue
I gathered data in Screen mode and VR mode: the number of frames and simulation time are as follows. 
[image]
As you see in screen mode the simulator updates around 3 times faster than VR mode.
Also, the root and environment in VR and Screen mode are same and the length of the road is about 2741 m having 45 seconds different is kind of weird!
Could you help me to improve my VR mode?
Thanks

ajdroid, 2023-10-04:
The VR mode performance depends primarily on the GPU available to you and which mode you're using (are you launching the sim via `make launch` or `make package` then `CarlaUE4.exe -vr`

closed 2024-01-05 by ajdroid