HARPLab/DReyeVR #140: Carla 0.9.13 installation issues on Windows 11

StijnOosterlinck, 2023-10-05:
Hi,

I'm having some difficulties trying to install Carla 0.9.13 on Windows 11.

I have all the prerequisites installed, but when trying to run 'make PythonAPI', I get an error.
Since the output is very large, I have put it in a file (the errors are displayed at the bottom):
[carla_log.txt](https://github.com/HARPLab/DReyeVR/files/12821266/carla_log.txt)

This is one of the errors I'm receiving: fatal error C1083: Cannot open include file: 'proj_api.h': No such file or directory (compiling source file C:\joan\carla\Build\om2odr-source\src\netbuild\NBHeightMapper.cpp) [C:\joan\carla\Build\osm2odr-visualstudio\src\osm2odr.vcxproj]

and also: C:\joan\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: Cannot open include file: 'OSM2ODR.h': No such file or directory

I'm not sure why these errors suddenly appeared because previously I was not getting those.
The only thing I can think of is that these appeared after I was trying to resolve another issue where I executed this command: 
`conda install -c anaconda xerces-c`

Is there anyone who has experienced similar issues and can help me with this?

StijnOosterlinck, 2023-10-07:
Update: this issue was related to tar not being able to extract proj-7.2.1.tar.gz properly, I temporarily fixed this by using a proj-install folder which I already had.

I still have another issue. I'm running the `make PythonAPI` command in an active conda environment using python 3.7.2, but it seems like python 3.11 is being used while compiling:
[image]

I would expect that instead of the '-IC:\Python\Python311\Include', my conda environment (with python 3.7.2 installed) would be shown there....

remark: executing `python -V` gives python 3.7.2, while executing `py -V` gives python 3.11.

Anyone knows how to fix this issue?

ajdroid, 2023-10-12:
We have not tested on Win 11, only on Win 10.
You can try changing the variable `py` in your Util/BuildTools/BuildPythonAPI.bat

closed 2024-01-05 by ajdroid