HARPLab/DReyeVR #142: eye tracking

FBanani, 2023-10-11:
Hi @ajdroid 

I understand my eye tracking has problems. 
First issue is about installation of "[VIVE_SRanipalInstaller_1.3.2.0.msi](https://hub.vive.com/en-US/download/VIVE_SRanipalInstaller_1.3.2.0.msi)". I found the solution. but the problem is that the color of robot eyes doesn't change, and it is gray!
what should I do?
the second one I found out during my installation I used the wrong SDK version "1.3.6.6" I downloaded the correct version and put it on the Carla toot/unreal/plugins. what should I do for the next step. Should I uninstall the DreyeVr and reinstall again? could you help me?
I really need help. I made 6 different maps for my experiment I don't want to uninstall Carla. what should I do?

ajdroid, 2023-10-11:
What exact VR headset model are you using?

FBanani, 2023-10-11:
> What exact VR headset model are you using?
> […](#)

thanks for replying. HTC VIV PRO 2

ajdroid, 2023-10-12:
We have not tested that headset but you should be able to get it to work
with some hacking. Can you use it with steam VR and calibrate the eye
tracking via the steam interface?

FBanani, 2023-10-12:
this is the exact headset that shows n your github 
[image]

ajdroid, 2023-10-12:
As mentioned in our README, we use the HTC Vive Pro Eye
[image]

ajdroid, 2023-10-12:
I'm not even sure that the Pro2 has eye tracking built-in

FBanani, 2023-10-12:
it has eye tracking but while I try to do it showed me initialization failed.

FBanani, 2023-10-12:
> I'm not even sure that the Pro2 has eye tracking built-in

[image]

ajdroid, 2023-10-12:
If you can find the SDK for enabling Vive Pro 2 eye tracking, you can use that and integrate it with DReyeVR. It doesn't look like the VIVE Pro 2 is supported by SRanipal (so we can't support it out of the box)

FBanani, 2023-10-12:
I try to fix this issue, because I ordered the HTC VIVE PRO eye tracking and I found out they sent me wrong product. I have another question, which is about the SDK version. Should I reinstall the DReyeVR again or it works just with putting the correct version in carla root/unreal/... directory?

ajdroid, 2023-10-12:
Do you mean for the Vive Pro Eye? If you use the supported Vive Pro Eye, you don't need to change anything -- just installing the SRanipal as described in the Install instructions will work. However, if you are trying to use a new SDK for the VIVE Pro 2, that is a much more involved effort and you'll need to change the DReyeVR source code to support it.

FBanani, 2023-10-12:
Yes, for Vive Pro Eye. what I understand is just putting the plugin in the root of carla. "unzip the SRanipal unreal plugin and copy over the 03_Unreal/Plugins/SRanipal/ directory to the Carla installation" after that what should I do?

ajdroid, 2023-10-12:
You also need to install the `.msi` file but I think you already did that. You don't need to do anything else

FBanani, 2023-10-12:
Thanks. to confirm, if I just had new headset, and put the file in a correct directory. I could do my experiment without doing anything els. am I correct?

closed 2024-01-05 by ajdroid