HARPLab/DReyeVR #143: Save 2D gaze position on the camera image

ALEX95GOGO, 2023-10-14:
Hello, I am struggling to calculate the 2D gaze position on the camera image, which should be the center of the reticle on the spectator screen. Is there a way to save the 2D gaze position variable? Or could you provide the formula for converting 3D gaze to 2D?

ajdroid, 2023-10-18:
You can see how it is done in C++ here: https://github.com/HARPLab/DReyeVR/blob/57109001edcfe0d91a6f814cf9c3287ff2bb57e4/DReyeVR/DReyeVRPawn.cpp#L227C1-L228C2

wx983, 2023-11-12:
@ALEX95GOGO I encountered the same issue with you. Do you solve it now?
@ajdroid  I don't know what to do next, could you please give more guides? 
Thanks!

ALEX95GOGO, 2023-11-15:
> You can see how it is done in C++ here: https://github.com/HARPLab/DReyeVR/blob/57109001edcfe0d91a6f814cf9c3287ff2bb57e4/DReyeVR/DReyeVRPawn.cpp#L227C1-L228C2

I managed to save the 2d gaze position by saving the ReticlePos variable in the C++ code and then recompile, here is my implementation:
/// HACK: correct for offset likely due to using left eye for camera projection
        ReticlePos.X += 0.3f * ReticleSize;
        std::ofstream outputFile("C:/Users/11550/Documents/carla/reticle_position.txt", std::ios::app); // Use std::ios::app to append to the file
        if (outputFile.is_open())
        {
            outputFile << "ReticlePos X, " << ReticlePos.X << ", ";;
            outputFile << "ReticlePos Y, " << ReticlePos.Y << ", ";;
            outputFile << "Time," << UGameplayStatics::GetRealTimeSeconds(World) << std::endl;
            //outputFile.close();
        }

ALEX95GOGO, 2023-11-15:
> @ALEX95GOGO I encountered the same issue with you. Do you solve it now? @ajdroid I don't know what to do next, could you please give more guides? Thanks!

I managed to save the 2d gaze position by saving the ReticlePos variable in the C++ code [](https://github.com/HARPLab/DReyeVR/blob/57109001edcfe0d91a6f814cf9c3287ff2bb57e4/DReyeVR/DReyeVRPawn.cpp#L227C1-L228C2)
and then recompile, here is my implementation:
/// HACK: correct for offset likely due to using left eye for camera projection
        ReticlePos.X += 0.3f * ReticleSize;
        std::ofstream outputFile("C:/Users/11550/Documents/carla/reticle_position.txt", std::ios::app); // Use std::ios::app to append to the file
        if (outputFile.is_open())
        {
            outputFile << "ReticlePos X, " << ReticlePos.X << ", ";;
            outputFile << "ReticlePos Y, " << ReticlePos.Y << ", ";;
            outputFile << "Time," << UGameplayStatics::GetRealTimeSeconds(World) << std::endl;
            //outputFile.close();
        }

wx983, 2023-11-27:
> I managed to save the 2d gaze position by saving the ReticlePos variable in the C++ code and then recompile, here is my implementation: /// HACK: correct for offset likely due to using left eye for [...]

Could you explain further what `recompile` is ? I mean, like `make pythonAPI`, `make launch` and `make package` to make changes of code to come into effects?

And another question, after this change as follows,
/// HACK: correct for offset likely due to using left eye for camera projection
ReticlePos.X += 0.3f * ReticleSize;
std::ofstream outputFile("C:/Users/11550/Documents/carla/reticle_position.txt", std::ios::app); // Use std::ios::app to append to the file
if (outputFile.is_open())
{
outputFile << "ReticlePos X, " << ReticlePos.X << ", ";;
outputFile << "ReticlePos Y, " << ReticlePos.Y << ", ";;
outputFile << "Time," << UGameplayStatics::GetRealTimeSeconds(World) << std::endl;
//outputFile.close();
}
Each time you open DReyeVR without extra pythonAPI programme, does it always generate a `reticle_position.txt` in the specific path?

ALEX95GOGO, 2023-11-27:
Yes, you have to recomplie. It has nothing to do with the PythonAPI, you can run make launch to test it and then make package. And for your second question, yes it will write lines into the txt file whe you run CarlaUE4.exe

> > > @ALEX95GOGO I encountered the same issue with you. Do you solve it now? @ajdroid I don't know what to do next, could you please give more guides? Thanks!
> > 
> > 
> > I managed to save the 2d gaze position by saving the ReticlePos variable in the C++ code and then recompile, here is my implementation: /// HACK: correct for offset likely due to using left eye for camera projection ReticlePos.X += 0.3f * ReticleSize; std::ofstream outputFile("C:/Users/11550/Documents/carla/reticle_position.txt", std::ios::app); // Use std::ios::app to append to the file if (outputFile.is_open()) { outputFile << "ReticlePos X, " << ReticlePos.X << ", ";; outputFile << "ReticlePos Y, " << ReticlePos.Y << ", ";; outputFile << "Time," << UGameplayStatics::GetRealTimeSeconds(World) << std::endl; //outputFile.close(); }
> 
> Could you explain further what `recompile` is ? I mean, like `make pythonAPI`, `make launch` and `make package` to make changes of code to come into effects?
> 
> And another question, after this change you mentioned as follows, /// HACK: correct for offset likely due to using left eye for camera projection ReticlePos.X += 0.3f * ReticleSize; std::ofstream outputFile("C:/Users/11550/Documents/carla/reticle_position.txt", std::ios::app); // Use std::ios::app to append to the file if (outputFile.is_open()) { outputFile << "ReticlePos X, " << ReticlePos.X << ", ";; outputFile << "ReticlePos Y, " << ReticlePos.Y << ", ";; outputFile << "Time," << UGameplayStatics::GetRealTimeSeconds(World) << std::endl; //outputFile.close(); } Each time you open DReyeVR without extra pythonAPI programme, does it always generate a `reticle_position.txt` in the specific path?
> 
> I will appreciate it a lot!

closed 2024-01-05 by ajdroid