HARPLab/DReyeVR #148: Custom spawn point

StijnOosterlinck, 2023-11-21:
Hi,

How can i set a custom spawn point when spawning the DReyeVR vehicle?

I want to use this when I only need a certain section of a map (CARLA map or own map). Spawning the vehicle at a desired location is then more efficient than driving there every time from the default spawn point, but I can not find where exactly the spawn point is defined...

Thanks!

ajdroid, 2023-11-27:
Hi -- you can achieve this by using [scenario runner ](https://github.com/carla-simulator/scenario_runner) (a core CARLA functionality) to define routes which start at your desired spawn point.

StijnOosterlinck, 2023-12-02:
Thanks for the answer @ajdroid , but i'm not looking for specific routes, since the participant is allowed to 'freely' drive around.

It seems logical that a map would have a default spawn point, or is this defined when making the map in roadrunner?
If so, I would just want to change the default spawn point.

ajdroid, 2023-12-02:
For Town 3 we specify spawn point in the DReyeVRConfig file. Otherwise we pick from spawn points in the map (see GetSpawnPoint in DReyeVRGameMode)

closed 2024-01-05 by ajdroid