HARPLab/DReyeVR #150: Blueprint scripting with DReyeVR

StijnOosterlinck, 2023-11-30:
Hi,

I was wondering whether we can use blueprint scripting together with DReyeVR?

I was especially looking for the functionality of "input keys" (from keyboard/logitech wheel).

Currently, I'm able to display a pause menu after a certain time, and remove it again after a certain time.
However, ideally, the pause menu should only be removed when a button on the logitech wheel is pressed (e.g. the 'A' key).
But using key input events seems not to work (also not with normal keyboard keys), is this because DReyeVR has already set bindings for all these keys?

I would like to give some functionality to some keys using the blueprints, and for this I'm also supposed to enable a setting called 'auto receive input' for a certain actor. For which actor should I set this then?

The scripts I am running now are just made in the blueprint class of the TeslaM3, but enabling this option here has no effect.

Has anyone experience with this?
Thanks in advance!

ajdroid, 2023-12-04:
Combining blueprint with C++ is doable but kinda messy and we don't recommend it. Instead you should try using the input handling logic we have in dreyevr pawn which does work and should be easily extendable so everything can be done in C++

ajdroid, 2024-01-05:
Sorry this may be possible but I have never tried to do it and am not sure if it will work

closed 2024-01-05 by ajdroid