HARPLab/DReyeVR #152: Cannot see changes the blueprint changes in pygame window

chirudeep001, 2024-01-02:
i have modified the blueprints for sounds and side mirror in the unreal editor and i could not see the chnages in the pygame window. Is there any specific thing to do, to see changes while i am running pygame window.

ajdroid, 2024-01-03:
Is there a reason you are using pygame? We generally use the unreal editor or the packaged build

chirudeep001, 2024-01-03:
I am using pygame for scenario runner.I  have done the changes in the unreal editor and built the package even though I could not see any changes in the pygame window.

Is there any setting to check so that the blueprints edited in the unreal editor reflects to the package when I built it??

ajdroid, 2024-01-03:
You don't need to use pygame for scenario runner. Running the editor and hitting play or using the packaged mode (ofc you have to redo make package) should work.

chirudeep001, 2024-01-03:
Sure , I will check on that and get back to you if I face difficulties.

Thank you

chirudeep001, 2024-01-05:
Hi, how are you running the manual control script and scenario runner.

1. firstly we need server and client to run the above scripts.
2. which client window are you using??
3. as soon as in run manual  control script pygame window opens up but thats not the case with your project, i believe thats the reason why i am not able to see changes made in the unreal editor not reflecting.

couild you help me with this. thank you

ajdroid, 2024-01-05:
DReyeVR works a bit differently than CARLA. We use the server directly as the "game window". It will spawn an ego vehicle and respond to keyborad/logitech wheel commands directly. No need to spwan an additional pygame window.

chirudeep001, 2024-01-16:
Small question regarding the fps, what was your FPS range and if it's higher than 25 how did you achieve that.

Thank you

chirudeep001, 2024-01-22:
Hi How to run the manual control script within the server window. Usually it opens the pygame window as soon as i run the manual control script. can you help me with this.

By default it uses pygame window (manual script from the git hub repo). 

Thank you
chirudeep

ajdroid, 2024-01-22:
You do not need to use manual control to do manual control. Simply run the simulator and you will be able to control the vehicle with your keyboard

chirudeep001, 2024-02-02:
Hi @ajdroid  

I am not using VR and also my question is in general what are the things to be modified to use carla server window as client window to display the sensor data (as your repo works) . for now i want to use scenario runner that runs in the carla server window. Which specific part of the code (in carla) should i work on to make this possible.

Thank you.