HARPLab/DReyeVR #154: Version mismatch between client and simulator API

Wingspear, 2024-01-19:
I'm trying to run some examples under PythonAPI. I'm currently launching the simulator via "make launch" and running the scene through Unreal. Then, I'm running some example python scripts via the client API.

Even though my python client seems to be on 0.9.13 (as it was pulled from the right commit), the client is still interpreted as 0.9.14. Added some screenshots below. Any pointers?

[image]

[image]

shh1v, 2024-01-20:
Can you list the files in your `$CARLA_ROOT$/PythonAPI/carla/dist` directory?

Wingspear, 2024-02-11:
hmm.. I don't seem to have a $CARLA_ROOT$/PythonAPI/carla/dist directory listed.
[image]

shh1v, 2024-02-11:
You are doing something wrong. The `make PythonAPI` should generate the egg and whl files. Are you running that command? Read the docs for installation.

Wingspear, 2024-02-11:
After trying `make PythonAPI` again, the output logs that "the dist folder was installed successfully", but it doesn't look like the installation went through. It appears to be a linker error when trying to find the Boost library (we're using Python 3.8). I attached some screenshots below. Any ideas?

[image]
[image]

shh1v, 2024-02-11:
Yes, if there are linker errors when you first run `make PythonAPI`, rerunning the command shows success; however, those errors are still there. I am familiar with these errors, please go through #46, and run the listed commands.

Make sure to _first_ run `make clean` and get rid of **all** the files in your build folder.

Wingspear, 2024-02-11:
I followed the instructions and reran `make PythonAPI`. After launching Unreal with `make launch` and running the `python generate_traffic.py` example, I'm still getting the same Client-Simulator version mismatch issue.
This is what my dist directory looks like:
[image]

shh1v, 2024-02-11:
did you reinstall the new python whl file using `pip install carla-0.9.13...whl` file? Did you first run `pip uninstall carla`?

Wingspear, 2024-02-11:
Awesome, `pip install carla-0.9.13...whl` did the trick. Thanks!

closed 2024-02-11 by Wingspear