HARPLab/DReyeVR #156: Changing spawnpoint via configfile works in packaged mode but not in editor

StijnOosterlinck, 2024-02-26:
Hi,

So the DReyeVRConfig file is meant to set parameters which could be changed at runtime without having to compile everything.
I was trying to do this for the SpawnEgoVehicleTransform, and lost some time because I thought it didn't work.

If I work with the editor, then I have to redo 'make launch' for the changes to take effect. However, I just noticed that when working with the packaged mode, changing the spawnTransform has effect without having to repackage the game.

Is there a specific reason why changing the spawnpoint for the editor mode requires recompilation?

ajdroid, 2024-02-26:
This is just an Unreal limitation/bug -- the editor does not reload the config for some reason. There are some threads about it such as [this](https://forums.unrealengine.com/t/how-to-reload-config-files-modified-in-the-editor/327925).
I mainly work with the package mode for re-editing config stuff. If you need the editor for something you can have that open as well, just don't "Play" it (use package mode instead)

StijnOosterlinck, 2024-02-26:
Oh okay, weird indeed, because some things (like setting the weather) are changeable via configfiles in the editor.

But i'll use the packaged mode then, thanks for the quick reply!