HARPLab/DReyeVR #178: How to fix a bug when setting up DReyeVR in VR mode for foggy

Victorrsl, 2025-09-08:
When I set up a foggy environment in VR mode, the air around me did not fog up, visibility was good and the sky was black. I've attached below a comparison picture between VR mode and non-VR mode. Can anyone help me analyze what the problem is and how I should fix this [bug.

[image]
[image]

.

Victorrsl, 2025-09-20:
This issue has been addressed in [#45](https://github.com/HARPLab/DReyeVR/issues/45).
Thank you, I'm glad to see your issue resolved. I'm wondering if the project needs to be recompiled? Or is it just a matter of changing the UE4 settings? I'm in a bit of a rush with this project right now, and compiling and installing DReyeVR was quite a hassle last time. I'm a bit apprehensive.

shh1v, 2025-09-21:
If you are working on takeover requests, you can have a look at my past two projects.
[NDRRI-Sim](https://github.com/[shh1v/NDRRI-Simulator](https://github.com/shh1v/NDRRI-Simulator)): Just an extension of CARLA but with the human-takeover request pipeline already implemented
[AutoHive](https://github.com/[shh1v/AutoHive](https://github.com/shh1v/AutoHive)): Also an extension of DReyeVR, but with all the scenario runner compatibility (with DReyeVR) bugs fixed and a fully configurable traffic TOR scenario implemented.

Below are the slides for setting up these projects on top of DReyeVR.
[Simulator Troubleshooting Slides.pdf](https://github.com/user-attachments/files/22449166/Simulator.Troubleshooting.Slides.pdf)

Good Luck!

Victorrsl, 2025-09-21:
Wow! Great results, thank you so much for sharing. I'll keep up the enthusiasm and follow this research direction. I've discovered that Volumetric Fog is the key to solving fog display, but I've also encountered a tricky problem: I can't get it to run stably on the Town04 map. It keeps throwing an error:“LowLevelFatalError [File:Unknown] [Line: 258]
Result failed
at
F:/UnrealEngine/Engine/Source/Runtime/Windows/D3D11RHI/Private/D3D11Que
ry.cpp:356
with error DXGI_ERROR_INVALID_CALL"

[image].
To resolve the above issues, I reset Vulkan as the default renderer, which resulted in errors occurring on any map (Town03/04/05):"The global shader cache file
"F:/CarlaDReyeVR/carla/Build/UE4Carla/0.9.13-dirty/WindowsNoEditor/Engine/Glo
balShaderCache-SF_VULKAN_ SM5.bin' is missing.
Your application is built to load COOKED content. No COOKED content was
found; This usually means you did not cook content for this build.
It also may indicate missing cooked data for a shader platform(e.g., OpenGL under
Windows): Make sure your platform's packaging settings include this Targeted
RHI.
Alternatively build and run the UNCOOKED version instead."

[image].
My computer hardware configuration:
Processor: 12th Gen Intel® Core™ i7-12700H 2.30 GHz
RAM: 24.0 GB (23.7 GB available)
Storage: 932 GB SSD Samsung SSD 980 PRO 1TB, 477 GB SSD SAMSUNG MZVL2512HCJQ-00BH1
Graphics Card    NVIDIA GeForce RTX 3050 Ti Laptop GPU (4 GB), Intel® Iris® Xe Graphics (128 MB) Do you have any experience resolving this issue? I would greatly appreciate your guidance.

Victorrsl, 2025-09-21:
After switching back to the following settings when I abandoned Vulkan, the PC version of Carla UE4 kept flickering and stuttering in VR mode, while the VR display remained normal. Previously, with Volumetric Fog enabled, it ran smoothly on the Town03/05 map. However, after setting Vulkan as the default to use the Town04 map, the situation worsened, and I couldn't revert to Carla's original stable state.setting like this：“[/Script/WindowsTargetPlatform.WindowsTargetSettings]
DefaultGraphicsRHI=DefaultGraphicsRHI_DX11
+TargetedRHIs=PCD3D_SM5

Victorrsl, 2025-09-22:
This issue has been resolved.